ok2cqr/cqrlog #18: HamLib fails to connect to rigctl

Efpophis, 2015-04-27:
I didn't know how else to let you know other than opening an issue ..

I tried out your hamlib implementation with my TS-2000, and it basically didn't work at all. So I added some debug code, which I've pushed to my fork here: https://github.com/Efpophis/cqrlog/tree/wk2x_fixing

The upshot is that the TCP connection is failing because connection is refused. This makes no sense, though, because I can establish as many connections to hamlib as I wish by using telnet or nc or whatever, and I'm not sure now how to proceed. I can keep trying in my limited free time, but I thought you should know as well - you should be able to test this to some extent, even without CW keying support on your radio.

ok2cqr, 2016-04-17:
It should be fixed in new version. Please try it and let me know. Thanks!

Efpophis, 2016-04-18:
Ah, yes - this has been fixed in 1.9.1 for a long time, and I have been using the feature with great success. This issue can certainly be closed.

closed 2016-04-18 by Efpophis