ok2cqr/cqrlog #34: mysql57dyn missing when building

kevinhooke, 2016-04-14:
I'm trying to build from source on Raspbian. I've installed the deps that are mentioned in your readme:
sudo apt-get install lazarus
sudo apt-get install libmysqld-dev
sudo apt-get install mysql-server

But I'm getting this error when I run make:
dLogUpload.pas(11,38) Fatal: Can't find unit mysql57dyn used by dLogUpload
ERROR: failed compiling of project /home/pi/radio/cqrlog/src/cqrlog.lpi

What do I need to install to get mysql57dyn?

Thanks!

ok2cqr, 2016-04-17:
You probably have old version of FreePascal compiler. Please install fpc version 3.0 and Lazarus 1.6 to build CQRLOG. If you are Ubuntu user, you can use my ppa on Launchpad.

73 Petr

closed 2016-04-19 by ok2cqr

kevinhooke, 2016-04-28:
Just fyi - I don't know why this didn't work for me last time (why I was trying to compile from source), but sudo apt-get install cqrlog worked on the latest Raspbian on a Pi3. Works great - thanks!