ok2cqr/cqrlog #35: Rigctl fails if instance of rigctld is already running

Efpophis, 2016-04-20:
If I start CQRlog "clean," meaning without rigctld running, it starts up and works fine. If I exit CQRlog, but do not log out or do anything else to manually kill the rigctld process, then the next time I start CQRlog it will fail to connect to rigctld and end up in a state with no ability to control the rig, read the frequency / mode, or send CW.  Steps to reproduce follow:

1. run ps -x | grep rigctld to confirm that rigctld is not running.
2. start cqrlog in debug mode - note that rigctl starts and everything is good
3. exit cqrlog
4. run ps -x | grep rigctld again to confirm that rigctld is still running .. it will be, but I don't think it should be - I think the child process isn't getting killed properly when the program exits.
5. start cqrlog again in debug mode - note that rigctl start fails and rig control is impossible.
6. run ps -x | grep rigctld to confirm that rigctld is running.
7. exit cqrlog
8. run ps -x | grep rigctld to confirm that rigctld is still running.
9. run pkill rigctld to kill rigctld manually, then ps -x | grep rigctld to confirm it's gone
10. start cqrlog in debug mode, note that rigctld starts and everything is good.

This is a new issue in version 2.0.0 (001).  I did not have this issue in the previous release.

As always, I'm happy to help however I can in tracking this down, testing, whatever.

ok2cqr, 2016-04-21:
What distribution are you using? I've never had this problem. The rigctld is always closed without any problem.

Efpophis, 2016-04-21:
I'm on ubuntu 17.2.

I took a look at the code that I think is supposed to kill the child process, and I notice there is supposed to be a "la" or some kind of debug output - I am not seeing this on the console when I run and quit in debug mode, either. It seems as if for some reason the destructor isn't being called, but I can't figure out why.

Efpophis, 2016-04-22:
I did some more debugging tonight, and I can confirm that the TRigCtl destructor code is never called in my case. One thing is that I am not opening the TRXControl window/form/whatever. I tried fiddling around to see if I could force something to call it on exit, but no luck.

closed 2016-05-05 by ok2cqr