ok2cqr/cqrlog #40: Exported ADIF file header should not start with "<"

aknrdureegaesr, 2016-09-23:
When exporting to an ADIF file, cqrlog currently exports a header such as

```
<ADIF_VER:5>2.2.1
ADIF export from CQRLOG for Linux version 2.0.2 (001)
Copyright (C) 2016 by Petr, OK2CQR and Martin, OK1RR

Internet: http://www.cqrlog.com

<PROGRAMID:6>CQRLOG
<PROGRAMVERSION:11>2.0.2 (001)
<EOH>
```

The ADIF spec says about headers:

> Optional header information may be included before the actual data in the file. To include optional header info, the first character of the file must be something other than &lt;.

Oddly enough, the spec for ADIF version 2.2.1 (for which the cqrlog header claims support) is not available on `adif.org`, but the above quote can be found in both ADIF spec versions [2.2.0](http://adif.org/adif220.htm) and  [2.2.2](http://www.adif.org/adif222.htm).

The current [spec version 3.0.4](http://adif.org/304/ADIF_304.htm) no longer contains this percise wording, but also rules out headers starting with &lt;:

> #### ADI File Structure
>
> An ADI file begins with an optional Header followed by one or more Records:
> ...
> #### ADI Header
> 
> A Header begins with any character other than &lt; and terminates with a case-insensitive End-Of-Header tag:
>
> &lt;EOH>    
> If the first character in an ADI file is <, it contains no Header. 

Suggested fix: Move the `<ADIF_VER:5>2.2.1` down.

aknrdureegaesr, 2016-09-23:
This is a duplicate, it was already reported earlier as https://www.cqrlog.com/node/1458 .

n1ywb, 2016-09-29:
It's still present in 2.0.1 at least; here are the top two lines 

`<ADIF_VER:5>2.2.1
ADIF export from CQRLOG for Linux version 2.0.1 (001)
`

n1ywb, 2016-09-29:
Did I mention I hate ADIF?

ok2cqr, 2016-10-04:
I'm sorry, I forget to fix the bug. I'll do it soon. Thanks!

commit cd86ea91 (2016-10-13): ADIF file cannot start with "<" character.
Fixes #40
closed 2016-10-13 by ok2cqr