ok2cqr/cqrlog #110: AppImage does not run

probonopd, 2018-07-25:
The AppImage does not run on Ubuntu 14.04 in my automated tests.

These libraries should not be bundled inside AppImages since they can prevent the application from running correctly:

```
WARNING: Blacklisted file ld-linux-x86-64.so.2 found
WARNING: Blacklisted file libc.so.6 found
WARNING: Blacklisted file libdl.so.2 found
WARNING: Blacklisted file libm.so.6 found
WARNING: Blacklisted file libpthread.so.0 found
WARNING: Blacklisted file librt.so.1 found
WARNING: Blacklisted file libstdc++.so.6 found
WARNING: Blacklisted file libGL.so.1 found
WARNING: Blacklisted file libdrm.so.2 found
WARNING: Blacklisted file libglapi.so.0 found
WARNING: Blacklisted file libxcb.so.1 found
WARNING: Blacklisted file libX11.so.6 found
WARNING: Blacklisted file libfontconfig.so.1 found
WARNING: Blacklisted file libfreetype.so.6 found
WARNING: Blacklisted file libexpat.so.1 found
WARNING: Blacklisted file libgcc_s.so.1 found
WARNING: Blacklisted file libglib-2.0.so.0 found
WARNING: Blacklisted file libICE.so.6 found
WARNING: Blacklisted file libSM.so.6 found
WARNING: Blacklisted file libuuid.so.1 found
WARNING: Blacklisted file libz.so.1 found
WARNING: Blacklisted file libgobject-2.0.so.0 found
```

How did you build this AppImage?

Reference:
https://github.com/AppImage/appimage.github.io/pull/527

ok2cqr, 2018-08-08:
I'm sorry, I'm not so keen in app image creation. I used the wiki and created the image. The libraries in the appimage are collected with simple script. Unfortunately I don't have it anymore, it was in testing VM and I already deleted it because the appimage could't be run on Fedora or newer Ubuntu. It always ended with core dumped and that's all. If you could help me to create the appimage of CQRLOG, it would be great. Thanks!

probonopd, 2018-08-08:
Most likely it was compiled on a too new system. Please compile on something like Ubuntu 14.04.

ok2cqr, 2018-08-11:
I'm sorry but I'm not sure if I can compile cqrlog under ubuntu 14.04 because it has too old lazarus and free pascal. Maybe GTK2 will work. Is there any script that will collect all needed libraries?

ok2cqr, 2018-08-11:
I just successfully compiled cqrlog under Ubuntu 14.04 using GTK2 version. Is there any way how to collect all needed libraries, please?

ok2cqr, 2018-08-12:
The application doesn't use QT when it's compiled in Ubuntu 14.04, I have to use GTK2.

probonopd, 2018-08-12:
`linuxdeployqt` is written in Qt but it can work on non-Qt apps, too.

ok2cqr, 2018-08-28:
I'm sorry but I'm giving this up. After 12 hours of my daily work I'm trying to create just simple app image without any extra dependecies like mysql, libhamlib, xplanet etc. I found documentation but it's tons of information but there is no simple step by step how to. I'm sorry, but I'm not interesting about any internal functions, not yet. I just wanted to create simple image from cqrlog.AppDir directory produced by make DESTDIR='cqrlog.AppDir' with a desktop file and icon. No way without reading MB of documentation and manuals how it works internally. Maybe I'm too tired for this. I'm sorry.

probonopd, 2018-08-28:
I can feel with you. How about this: You get the application to compile on Ubuntu 14.04 on Travis CI, and I do the rest and send you a pull request?

ok2cqr, 2018-08-29:
Travis CI is another problem. I can install deb packages of Lazarus and fpc to Ubuntu 14.04 but because of old libraries, I have to recompile Lazarus by hand before I can compile CQRLOG. Will that be possible with Travis CI?

probonopd, 2018-08-29:
> Travis CI is another problem. I can install deb packages of Lazarus and fpc to Ubuntu 14.04 but because of old libraries, I have to recompile Lazarus by hand before I can compile CQRLOG.

That has nothing to do with Travis CI itself, but with the fact that if you want your application to run on all supported Ubuntu LTS releases you need to compile on the oldest LTS release (which is currently Ubuntu trusty). This is how backward compatibility in general works. Like if you compile on Windows 7, the application will still run on 8 and 10, but not the other way around.

> Will that be possible with Travis CI?

Yes, why not.

ok2cqr, 2018-08-30:
I just finished testing the travis.yml file and the build works. It ends with successful compilation of cqrlog. What next, please? Thanks!

https://github.com/ok2cqr/cqrlog/blob/master/.travis.yml

probonopd, 2018-08-31:
is there no `configure` and no `install` step?

We need something like this.

In the case of CMake:

```
  - cmake . -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr
  - make -j$(nproc)
  - make DESTDIR=appdir -j$(nproc) install ; find appdir/
```

In the case of autotools:

````
  - ./configure --prefix=/usr
  - make -j$(nproc)
  - make install DESTDIR=$(readlink -f appdir) ; find appdir/
````

In the case of qmake:

```
  - qmake CONFIG+=release PREFIX=/usr
  - make -j$(nproc)
  - make INSTALL_ROOT=appdir -j$(nproc) install ; find appdir/
```

Can you add something like that? The net result is that the binary should have `/usr` as the compile-time prefix, but installation should happen to `./appdir/usr`.

ok2cqr, 2018-09-01:
There is no configure step. Only make and make install. I just added make install to the script and build was successful.

probonopd, 2018-09-02:
Initial AppImage is here https://github.com/probonopd/cqrlog/releases but it seems there are dependencies on `mysqld` and even on `libsql*` (using `dlopen()`?) which we need to handle by bundling them into the AppImage.

ok2cqr, 2018-09-02:
Wow, that is amazing! 

There a few other dependencies but program won't work without libmysql libraries because it has to connect to a database.

What is this, please?
    "URLS=$(sudo apt-get install --print-uris  mysql-server-core-5.5 | grep \"^'http\" | cut -d \"'\" -f 2)",
    "( cd appdir/ ; wget -c $URLS ; find . -name \"*.deb\" -exec dpkg -x {} . \\; ; rm *.deb )",

It's trying to install mysql server to be a part of app image? Could I use the same way to install other libraries like libhamlib2 and libhamlib-utils? It would be much better to use mysql-server-core-5.6 that is in the repo as well because it's impossible to upgrade to newer MySQL directly from 5.5.
It there any way how to debug the application when it's running from app image? I have some kind of debug mode adding --debug=1 as a parameter to cqrlog binary. It writes information to standard output. Thanks a lot!

probonopd, 2018-09-02:
> It's trying to install mysql server to be a part of app image?

Apparently this application needs the `mysqld` executable on the `$PATH`. Since this does not come with all Linux distributions in the default desktop installation, we need to bundle a suitable version of it inside the AppImage. This is what I am fighting with at the moment, because MySQL also wants to read stuff from locations like `/var/lib/mysql` which of course don't exist if no MySQL is installed on the host system.

> Could I use the same way to install other libraries like libhamlib2 and libhamlib-utils? 

If your application links to those, they should be bundled automatically. Or are you using `dlopen()`? Then we need to copy them in manually.

>  it's impossible to upgrade to newer MySQL directly from 5.5

Whatever MySQL goes into the AppImage is the version that will be used. No upgrading is possible or desired.

> I have some kind of debug mode adding --debug=1 as a parameter to cqrlog binary.

Then you should be able to use the same argument on the AppImage as arguments are simply passed through to the payload executable.

probonopd, 2018-09-02:
Would it be possible to remove the dependency on the `mysqld` executable?

ok2cqr, 2018-09-02:
When  you run the cqrlog for the first time, it asks if you want to store data into $HOME. If you click to yes, it starts own instance of MySQL server custom config file and with parameters to work as embedded database. 
https://github.com/ok2cqr/cqrlog/blob/master/src/dData.pas#L3427

It would be great to keep this feature in the application. It's much easier for users to have the data in the home directory, they haven't to enter any passwords etc.

Yes, the MySQL libraries are loaded with dlopen.

probonopd, 2018-09-02:
> It would be great to keep this feature in the application. It's much easier for users to have the data in the home directory, they haven't to enter any passwords etc.

Sure, that's good. 

What apparently is not so easy is to bundle MySQL itself in the AppImage, though.

> Yes, the MySQL libraries are loaded with dlopen.

From which location? We need a location relative to main binary.

ok2cqr, 2018-09-02:
It checks various locations
/usr/bin/mysqld
/usr/bin/mysqld_safe
/usr/sbin/mysqld

but I can modify it to check any other path. What the path should I check? The same also with libssl libraries, right?

ok2cqr, 2018-09-02:
I just added the path to the app executable directory.

Yes, I know, but still hope it will work :).

probonopd, 2018-09-02:
Using SQLite is not an option? (Would make things _much_ easier...)

dl8bh, 2018-09-03:
Me and some other guys are using MySQL to have the database public reachable (we connect to a MySQL Server other than localhost which also has a webinterface to log directly to the database). Switching to SQLite would make our approach almost impossible.

probonopd, 2018-09-03:
OK, I see. This means we'll need to get creative ;-)

ok2cqr, 2018-09-05:
The mysql-server as dependency could be removed but  program will lost the ability to save data to $HOME. It could not be so big problem but it still needs the libmysql client libraries. Unfortunately it will need multiple versions and even the mariadb client libraries because people can connect to MySQL or MariaDB depends on version of distribution there are using.

I've seen there are some PPA with MySQL 5.7 for Ubuntu 14.04, so installing all the client libraries shouldn't be the big problem.

probonopd, 2018-09-06:
https://bugs.mysql.com/bug.php?id=37697
5 Oct 2009 

> We hope to provide fully relocatable packages in a future release.

Wonder what came of it.

probonopd, 2018-09-06:
Yes, I think not needing `mysql` would already be a big plus (hence I was thinking about using SQLite as a possible alternative, but I understand that it's not that easy).

ok2cqr, 2018-09-08:
Unfortunately SQLlite is not an alternative. CQRLOG has several threads that work with database and even need write access. As far as I know, Sqlite doesn't support that. It can handle multiple queris reading the data but only one that writes. I could write some kind of queue but it won't solve problem with remote access to the database.

I just cloned clean version of Ubuntu Trusty and here is the list of commands to install all necessary mysql client libraries including MariaDB.

```bash
sudo apt-key adv --keyserver pgp.mit.edu --recv-keys 5072E1F5
sudo sh -c "echo 'deb http://repo.mysql.com/apt/ubuntu/ trusty mysql-5.7' >> /etc/apt/sources.list"
sudo sh -c "echo 'deb http://repo.mysql.com/apt/ubuntu/ trusty mysql-8.0' >> /etc/apt/sources.list"
sudo apt-key adv --recv-keys --keyserver hkp://keyserver.ubuntu.com:80 0xcbcb082a1bb943db 
sudo add-apt-repository 'deb http://ftp.osuosl.org/pub/mariadb/repo/10.0/ubuntu trusty main'
sudo apt-get update
sudo apt-get install software-properties-common
sudo apt-get install libmysqlclient18 libmysqlclient20 libmysqlclient21 mysql-common libmariadbclient18
```

Still it would be great to have mysql server part of the appimage, but if it is not possilbe, we can move it. I''l write some kind of manual for users how to use installed mysql version in the system.

I tried appimage you've created using travis and works very well when use external mysql server. Excellent work! Thanks!

probonopd, 2018-09-09:
> works very well when use external mysql server

Well, it's not really ideal since the user needs to do more than download and run just one file. But it's the best I can do at the moment without investing much more time on MySQL.

lucifer, 2019-05-17:
Just came across this project, looks excellent btw; would definitely benefit from being contained. 
I've no time to commit code on this yet but I'd probably use Docker and X11 for multi-platform spread. I'll be back (: 👍

ok2cqr, 2019-07-26:
It's possible to try to integrate mysql8.0 into appimage? We could use https://dev.mysql.com/downloads/mysql/ minimal install in tar.gz. I'd like to create the final appimage.
It's possible to edit travis,yml file on the travis website and run the build directly from it? It would be better than editing the file in CQRLOG repository. I did it somehow a year ago, but don't see the option on current website version :(.

probonopd, 2019-07-26:
Yes, that would be reasonable to do.

AsciiWolf, 2020-09-05:
What is the state of this issue?