ok2cqr/cqrlog #112: TMySQL57Connection can not work with the installed MySQL client version: Expected (5.7), got (6.1.11).

sq9fk, 2018-07-26:
Hello
I've latest MariaDB 10.1.34-MariaDB stable on Gentoo Linux
Acording to DEBUG info:
Loading libmysqlclient: /usr/lib64/libmysqlclient.so.18.4.0
**************************
MySQL version: 6.1
**************************
**********************************
MySQL version assigned: 6.1
**********************************

Could you please make a workaround for version checking in dData.pas:
Line 4167: procedure TdmData.CreateDBConnections;

Thank you.
Piotr SQ9FK

ok2cqr, 2018-08-08:
Hi Piotr,

it seems you have old version of libmysqlclient library, install the recent version and log should work.

73 Petr

sq9fk, 2018-08-08:
Hello Petr
6.1.11 is the latest and only available at gentoo repository.

So we should make change in source to have compatibility with stable latest gentoo release.

Piotr SQ9FK 

-- Wysłane za pomocą K-9 Mail.

HoracioDos, 2018-08-10:
Hello
I have a quite similar issue while connecting to a mysql-server version 8. 
TMySQL57Connection can not work with the installed MySQL client version: Expected (5.7), got (8.0.12).
If I try to use a local database I get this.
2018-08-10T00:47:43.912213Z 0 [System] [MY-010116] [Server] /usr/sbin/mysqld (mysqld 8.0.12) starting as process 10979
2018-08-10T00:47:43.919451Z 1 [ERROR] [MY-011011] [Server] Failed to find valid data directory.
2018-08-10T00:47:43.919646Z 0 [ERROR] [MY-010020] [Server] Data Dictionary initialization failed.
2018-08-10T00:47:43.919679Z 0 [ERROR] [MY-010119] [Server] Aborting
2018-08-10T00:47:43.920082Z 0 [System] [MY-010910] [Server] /usr/sbin/mysqld: Shutdown complete (mysqld 8.0.12)  MySQL Community Server - GPL.
Thanks!

HoracioDos, 2018-08-10:
Hello.
While trying to change source code I found out that there is no data component for mysql 8 in Lazarus and Zeoslib could provide access to version 8. Oracle support will continue up to Oct 2020/2023 for version 5.7 but many people will start using version 8. 
I read somewhere that it's possible to install mariadb alongside mysql but it seems very tricky. I know that in early versions of cqrlog you've tried to use sqllite without success due to a thread problem. Perhaps you could check it again. It's not a problem right now but it will in the near future.
Thanks!

gl-bars, 2018-12-05:
There is a ticket for adding mysql 8 component: https://bugs.freepascal.org/view.php?id=34257

lx2gt, 2020-01-30:
> Hello
> I have a quite similar issue while connecting to a mysql-server version 8.
> TMySQL57Connection can not work with the installed MySQL client version: Expected (5.7), got (8.0.12).
> If I try to [...]

Your Data directory is not initialized, MySQL8 will then not work. CQRLOG needs to do an mysqld --initialize on ~/.config/cqrlog/database

Luc

closed 2026-10-08 by ok2cqr