ok2cqr/cqrlog #170: Online log upload

dl7oap, 2019-05-10:
the online log upload function seems to stop, when 
- it tries to delete a qso, which is not in the in the online log
- the online log service is not online/shut down

the online log function should not stop, but
- continue with the next qso upload/change/delete
- continue with the other online log services (some user uploads parallel to hamqth/hrdlog/clublog)
- log errors which can not been upload/change/delete in a file, so user can take care for the qsos which have problems

dl7oap, 2019-06-12:
more hints to this error  https://www.cqrlog.com/node/2195

phl0, 2020-05-27:
@dl7oap Could you please test this?

https://github.com/phl0/cqrlog/commit/60af4fd3e384070698ac058289a6e5111d25f121

Not sure if that breaks something else but at least it continues in case in cannot upload/delete a QSO.

phl0, 2020-07-07:
I made a new approach solving this issue:

https://github.com/ok2cqr/cqrlog/compare/master...phl0:moreDetailedErrorHandlingForOnlineLogs

Basically FatalError was changed to an integer variable being 0 for no errors, 1 for fatal error (upload completely stops) or 2 in case that the error is not fatal and upload should simply continue with the next QSO to upload,
@dl7oap Would you be able to verify my changes?

dl7oap, 2020-07-08:
Hi, i did a CodeReview and the Code looks fine for me.
I only use hamqth, so i can not test it in real life.

Bye dl7oap

phl0, 2020-07-09:
There is at least one point where it should continue with HamQTH upload:

[image]

But I think we cannot provoke this situation... :)

mcalistri, 2022-08-15:
This bug is still present on CQRLOG  2.6.0 for Clublog  and HRDLog as I highlighted on the CQRLOG Forum.

73's de PY1ZRJ