ok2cqr/cqrlog #178: support for 4K/UHD screens

dl7oap, 2019-06-24:
support for 4K/UHD screens
https://www.cqrlog.com/node/2262
cqrlog should be make ready to support 4K/UHD screens out of the box.

oliversturm, 2020-06-29:
I'm very interested in this. I don't use any computers anymore that have simple HD screens - even my laptop runs at a "high" DPI.  As far as I see, the issue with cqrlog is that it uses GTK2, which doesn't support scaling. I have tried using a theme created by https://github.com/themix-project/oomox - it has the ability of creating extra-large themes to fake High DPI for GTK2 - but cqrlog looked exactly the same with the theme applied.

I found a forum thread that mentioned QT5, which would obviously be an option. If there is such a build available, I'd be happy to test it. On the other hand, wouldn't it be easier to move to GTK3? 

What would it take to make a High DPI-cqrlog happen?

OH1KH, 2020-07-01:
If you can compile source you can try QT5. At the moment only known problem is that Grayline map does not connect to RBN with QT5. All other things seems to work so far.
If you can not complle look at my Git cqrlog clone, branch "loc_testing". There is subfolder "compiled" with QT5 executable and instructions how to update. https://github.com/OH1KH/cqrlog/tree/loc_testing/compiled
You can try it if you first install a working official version and then update (maybe to virtual machine, just for testing)

GTK3 does not compile cqrlog at the moment. It stops to some Lazarus internal code bugs. ( https://wiki.lazarus.freepascal.org/Qt5_Interface#Why_Qt5_instead_of_GTK_.3F )

oliversturm, 2020-07-01:
I made a test yesterday evening on my Chromebook and managed to compile with QT5. The result on the Chromebook was pretty good in terms of UI layout, but I noticed a few issues with buttons (couldn't click them - had to use the keyboard) and menus (very high items). However, I don't have any actual data on the Chromebook, so I will repeat the process on a machine that I normally use to run - there's a chance that things will work better there as well, since the Chromebook sometimes has X11/Wayland issues anyway. I'll update this thread when I have results.

oliversturm, 2020-07-01:
Okay, built my own now on the real machine and it works perfectly - no issues with buttons or menus at all. Very usable on the 4k screen as well. Excellent!

Btw - I left the existing installation in place on this machine and simply installed all the new stuff in `~/.local` and sub-folders. This way it's easy to remove or to replace with newer versions in the future.

OH1KH, 2020-07-02:
OK!
Nice to hear. Could we consider to close this issue because QT5 works for solution to 4K screens?

oliversturm, 2020-07-04:
Perhaps there should be an "official" package for the qt5 version so it becomes more accessible for those who don't want to install or build manually.

sq6emm, 2021-12-08:
Anything in this topic?:) cqrlog is still unusable on 4k/uhd screens;(

OH1KH, 2021-12-09:
Hi !
I do not work with packages nor have any 4k/uhd screens for testing ;(
If I understood right Petr has had some difficulties with packaging lately.

But perhaps new package could have qt5pas included in dependencies  and then have two binaries "cqrlog" and "cqrlog_qt" inside to choose.

That is just my quick idea...

sq6emm, 2021-12-09:
Is this about a lot of development work to confirm qt5 is working with UHD/4k screens? Or it is just about compilation with newer library and making tests?
If it is the second one than it makes sense for me to try...
Or maybe you already have some branch on which it can be tested by me for example?

oliversturm, 2021-12-09:
The version I built for myself back in July 2020 worked fine with QT5 and HD screens. I stopped using it for various reasons, but I would expect this approach to still work today. An included binary that uses QT5 should probably solve this problem for everybody.

sq6emm, 2021-12-09:
Yep I am tryin to get it working on XPS 13 that works with: 3840x2400 resolution with 200% scale.

sq6emm, 2021-12-09:
@oliversturm Thanks for your work on it...
I've fresh Ubuntu 20.04 LTS with all updates.
Installed libqt5pas1.
Overwritten your binary of cqrlog and also help files.
But I get Access violation on early begining;\

OH1KH, 2021-12-10:
Cloning or downloading source ZIP from GitHub includes Makefile that has option "cqrlog_qt5"

It has been there for some time now. Instead of issuing "make" just issue "make cqrlog_qt5" will do qt5 version.
just be sure that libqt5pas (or what name your linux distribution uses) and it's devel package are both installed.

Developing cqrlog is the same with GTK and QT5. It is just the compile stage that needs another option.
And of course checking that both versions results same way. Sometimes QT5 needs more attention to form design.
The results are depended on how FPC/Lazarus IDE does the job. There is nothing else than from design checks that we can do for QT5 version of cqrlog. 

What comes to my laptop, that is my main computer, it is just an old  second hand Lenovo ThinkPad X201 with 1280x800 resolution (it even does not have HDMI socket!) and secondary display with 1440x900 resolution.
I don't have glorious Hamshack here. See it at time 0:55 of this video https://www.youtube.com/watch?v=Vd-dhD8acIs&t=5s