ok2cqr/cqrlog #225: Optionally also include operator's call in exports

OH1KH, 2019-11-18:
I think we should follow standards with adif.

OPERATOR | String |   | the logging operator's callsign if STATION_CALLSIGN is absent, OPERATOR shall be treated as both the logging station's callsign and the logging operator's callsign.
 ( https://www.adif.org/310/ADIF_310.htm#QSO_Field_OPERATOR )

So we need to use preferences/station/call to tag STATION_CALLSIGN's data and found a place where we can fill operator callsign for tag OPERATOR if that is different.

Either in preferences/station or NewQSO must have "operator" filed for that.

Or have I understood wrong?

dl7oap, 2020-02-20:
in a contest the STATION_CALLSIGN ist the contest call. OPERATOR gets the call of the oldman behind the mic.

under normal circumstances  STATION_CALLSIGN and OPERATOR is the same. like you already have written preferences/station/call.

where to place operator info? 
- could be a new database filed in cqrlog_main
- could be a new database field in the profile? is a contest like a own profile ?
- could be a new database field in the notes? so information is only linked to a cqrlog_main, when its necessary?

there are many ways..

phl0, 2020-09-02:
Hej @dl7oap @OH1KH, I did some line of code to start with this topic. See https://github.com/phl0/cqrlog/tree/operator.
What do you think? Is this the way to go?

OH1KH, 2020-09-02:
Hi!
  Did not see any diff (yet?) between operator branch and master.

 - My vote is a database field.
 - Quick setting via NewQSO menu/ctrl+key. Visible at NewQSO status bar (where "My grid" is shown, too). 
 - Not resetting when qso is logged, but set always as station call when cqrlog starts. No need to have it in preferences.
 (At club: a member can work several qsos, so no need to set operator for every qso. When cqrlog is started is assumed that operator may be other member than last time was)
- wsjtx logging has operator field it could then be set to database field when logging. Should be set only in case:
 wsjtxoperator<>'' and cqrlogoperator=stationcall else operator from cqrlog's operator setting.
(At club: member needs to set only cqrlog operator, but he can leave it as is operator=stationcall and put operator to wsjtx side, as alternative, if he works only wsjtx qsos)
- then exported properly from database to all ADIF/HTML/EDI/CABRILLO

OH1KH, 2020-09-02:
UPD:
I think fldigi has also "operator" setting.
(Can not open it now as it is installed from package, but my Hamlib is compiled from source what makes libhamlib.so.2 loading error.)
But same rules for that as wsjtx.

phl0, 2020-09-02:
There are 3 commits ahead of the upstream master already.
See: https://github.com/ok2cqr/cqrlog/compare/master...phl0:operator

closed 2020-11-24 by OH1KH