ok2cqr/cqrlog #241: RBN connection Grayline map/QT5

OH1KH, 2020-01-16:
Connection to RBN via grayline map does not work if compiled for QT5.
Popup selector stays "Connect" (no change to "Disconnect"  and "Disconnected" disappears from statusbar leaving it clear.

Debug results:
Server:telnet.reversebeacon.net Port:7000
QSocketNotifier: Can only be used with threads started with QThread
QSocketNotifier: Can only be used with threads started with QThread
QSocketNotifier: Can only be used with threads started with QThread

When compiled for gtk2 works as should.

OH1KH, 2020-01-16:
As far as I understand I have Fedora 30 with LDXE and no Wayland. Tried with LXDE and OpenBox desktops but both have same error. It might be that in some Linuxes this might even work (???).

Found this from freepascal wiki:

Systems using Wayland
Some distributions us Wayland by default (eg the default Fedora with a GNOME desktop) and QT5 does not work with Wayland. You will receive a message like:

QSocketNotifier: Can only be used with threads started with QThread
[FORMS.PP] ExceptionOccurred
....
Logout, from the login screen click the small gear symbol, select "GNOME on XOrg". Log back in again.

ok2cqr, 2020-02-11:
I have the same problem on system without Wayland, just using QT5 version. It seems there is some diference in something related to threads, I have to figure out what.

OH1KH, 2020-02-11:
OK.
This s unresolved. I compared the connection with others and I can not see any big difference how connection is created, It is not a big thing anyway, but would be nice to get it work.
There was another QT5 issue somewhere, but forgot where. I always should make issue report immediately when something is found.
It is then easier to return on later time.

OH1KH, 2020-02-19:
There is also another issue with grayline map spotting. It is the same with gtk2 or qt5.
Set new qso callsign  ZL1NA/MM and the short path line is drawn to ZL. Even when NewQSo/DXCC ref. says: "!",  that is right as we do not know where /MM is located.

Keeping that in mind no path line should be drawn unless locator grid is also filled.

OH1KH, 2020-02-19:
I think I have now solution for /MM path line case.
When DXCC is changed to '!' path line is wiped. If after that a locator is entered path line is drawn again based on myloc/loc.

I must test I did not break anything else.

Still have to fix path line to 0,0 if random state is entered (exit from edtState) to empty NewQSO (edtCall = '').

About main problem Telnet connect with QT5 it may be because in grayline map form  the "lTelnet" is inside the type definition TRBNThread, but in rbnMonitor form "lTelnet" is placed to TfrmRbnMonitor type definition, not into TRbnThread (and so it works with QT5, perhaps).

It should be changed for testing, but may be too complicated for me.

OH1KH, 2020-11-24:
This is now (temporarily ?) fixed by commit #348. Closing issue.

closed 2020-11-24 by OH1KH