ok2cqr/cqrlog #298: TX Frequency in rig control to get proper ADIF logs for SAT use

kng, 2020-06-24:
Running rigctld 3.3 without --vfo mode, gpredict 2.3 controlling a FT-847 in SAT mode.
In this example uplink on 70cm and receiving on 2m, set for AO-91.

When cqrlog sends 'fmv' it gets the receive frequency and displays 145M, if you instead send 'imv' you get the TX frequency on 435M instead. It seems to also work OK when not in satellite mode. Probably needs to be tested on more rigs.
Perhaps also better to send 'ims' to see if the rig operates in split mode.

This is according to https://www.eqsl.cc/qslcard/SatelliteInfo.cfm
"Remember in ADIF crossband operation, the BAND is the transmission band, not the received band"

A short description of the replies from rigctl -m 2:
get_freq (f)
Frequency: 145960000

get_mode (m)
Mode: FM
Passband: 15000

get_vfo (v)
get_vfo: error = Feature not available

get_split_vfo (s)
Split: 1
TX VFO: Sub

get_split_freq (i)
TX Frequency: 435250000

//SA2KNG

OH1KH, 2020-06-24:
Hamlib 4.0 has lot of changes. All future development should be based on version hamlib 4 or up.
How ever at the moment hamlib itself is also in big change, so it might be better to wait a while that biggest bugs are fixed.

Proper way to know what rig can do is to send "\dump_caps" and use the result as base for commands.
How ever cqrlog's hamlib interface is very simple and limited.Best would be to write completely new unit for hamlib communication.

While hamlib should offer equal interface for high level programs it is not so simple.
A while ago I removed "v" from "fmv" as Icoms can not handle that anyway and no vfo information is used in cqrlog.
Few weeks later I got report that Kenwood rigs do not work properly if "v" is removed. So it is back there again.
Same with "imv". I suspect that it will cause problems with some rigs while others work ok.
That is why \dump_caps usage is important, but it makes the communicating unit lot more complex.

kng, 2020-07-06:
Perhaps letting the user choose what queries to send ? Some checkbox or dropdown to choose from a subset of those strings, the code seemed to parse "imv" responses without any modification. I've never used lazarus so messing with the gui is probably a bad idea.

OH1KH, 2020-07-06:
Maybe.
But if you need "imv" it is very easy to change with hexedit to /usr/bin/cqrlog.
look for hex string 66 6D 76 0A. Jump to ascii side with TAB and change f to i and save.
Should be found two times.