ok2cqr/cqrlog #305: EncodeURL on password insufficient for LoTW connection URLs

oliversturm, 2020-07-04:
I found that my LoTW upload and download were not working correctly - I received output in the dialog that looked like the HTML for the login page, including a message saying something like "username or password incorrect". I looked at the source and found that [the password is included in the target URL](https://github.com/ok2cqr/cqrlog/blob/e9b62e04283233ef49112e9355f0de5012a329a1/src/fImportLoTWWeb.pas#L133) (this one is for download) after running through `EncodeURL`. 

Some tests showed me that EncodeURL does not encode ampersand characters (`&`) in the password, resulting in a corrupt URL. I changed my password to one that doesn't include an ampersand and now everything works correctly.

OH1KH, 2020-07-04:
That is not true, function works as expected.
Clipped the function EncodeURLData from source and wrote small test program using that function.

Begin
  line :='ABCDEFGHIJKLMNOPQRSTUVWXYZ #$&+,/:;=?@[]abcdefghijklmnopqrstuvwxyz01234567890';
  writeln (line);
  writeln ( Form1.EncodeURLData(line));
end.          

Here is the run result:
ABCDEFGHIJKLMNOPQRSTUVWXYZ #$&+,/:;=?@[]abcdefghijklmnopqrstuvwxyz01234567890
ABCDEFGHIJKLMNOPQRSTUVWXYZ%20%23%24%26%2B%2C%2F%3A%3B%3D%3F@%5B%5Dabcdefghijklmnopqrstuvwxyz01234567890

From result you can see that ampersand is coded OK as %26. It is the LoTW server that extracts HTML get  request parameters separated by ampersand and can not make difference with %26 and printable &

dl7oap, 2020-07-04:
I can, too, not reproduce this error. i have, too, a @ and other special characters in my lotw password. and webdownload and the upload working fine with the newest hamlib version (last commit of master).

must be then a other problem (could it be a wrong iso of database of the user?)

OH1KH, 2020-07-04:
Ahhhh!

My fault !

LoTW part uses EncodeUrl function that is part of synacode unit.
How ever there is cqrlog's own dmUtils.EncodeUrlData function that I wrongly refered.

Synacode's encode can not encode ampersand, cqrlog's own can. (tested)
So this is a bug.

We must replace synacodes encode with cqrlog's encode in all places where html get is used.

Sorry!

dl7oap, 2020-07-04:
hm, here lotw download is still working with EncodeURL. without problems. 
i changed it to EncodeURLData, but this function, too, do not touch @, because this is in safemask array. which prevent that @ is changed to something other.

a 'ö' is changed, but this is, too, changed in EncodeURL here.

i cannot help here, because i can not reproduce this error. maybe i don't see, what i am doing wrong. using lazarus v.2.0.8 and fpc 3.0.4 with ubuntu 18.04

dl7oap, 2020-07-04:
Hi, yes EncodeURLData should be the better way. there are some more points in the code where EncodeURL is used. think we should change all to EncodeURLData.

OH1KH, 2020-08-23:
As fix merged oliversturm can you close  this issue now?

oliversturm, 2020-08-24:
Yes, fine with me 👍

closed 2020-08-24 by oliversturm