ok2cqr/cqrlog #309: TRX control not working / FT-DX101MP

oliversturm, 2020-07-25:
I just hooked up my new FT-DX101MP. I run rigctld from cqrlog and this was previously working correctly with an IC-9100. Now I have the strange situation that other applications can talk to the radio correctly (tried jtdx, for instance), but cqrlog can't control the radio in the TRX control window, or show me current state information.

I use model 1040 (FT-DX101D), serial speed set to 38400 (as configured on the radio), all other settings on "default".  When I start cqrlog, I see this output: 

```
...
Settings:
-----------------------------------------------------
RigCtldPath:/usr/bin/rigctld
RigCtldArgs:-m 1040 -r /dev/ttyUSB0 -t 4532  -s 38400 
RunRigCtld: TRUE
RigDevice:  /dev/ttyUSB0
RigCtldPort:4532
RigCtldHost:localhost
RigPoll:    500
RigSendCWR: FALSE
RigId:      1040

Starting RigCtld ...
rigProcess.Executable: /usr/bin/rigctld
Parameters:
-m
1040
-r
/dev/ttyUSB0
-t
4532
-s
38400

Recommend using --vfo switch for rigctld if client supports it
rigctl and netrigctl will automatically detect vfo mode
rigctld started!
Connected to rigctld @ localhost:4532
Sending: � 1

Sending: fmv

Sending: fmv

Sending: fmv

Sending: fmv

...
```

I'm not sure what the character is that is sent the very first time - it displays just like that on the console.

The weird thing is that when I try the same command manually, it works correctly:

```
> rigctl -m 1040 -r /dev/ttyUSB0 -s 38400 -vvvv

...
Opened rig model 1040, 'FT-DX101D'
Backend version: 20200306.0, Status: Beta

Rig command: fmv
rig_get_freq called
newcat_get_freq called
newcat_valid_command called
newcat_valid_command called
write_block called
Frequency: 7193070
VFO: VFOA

Rig command: rig_get_mode called
newcat_get_mode called
newcat_valid_command called
write_block called
rig_passband_normal called
newcat_get_rx_bandwidth called
newcat_valid_command called
write_block called
rig_passband_wide called
Mode: LSB
Passband: 3000

Rig command: rig_get_vfo called
newcat_get_vfo called
newcat_valid_command called
write_block called
newcat_get_vfo_mode called
newcat_valid_command called
write_block called
VFO: VFOA

...
```

So... what's going on? Apparently cqrlog doesn't receive a response it can understand? Is there some way to activate additional logging so I can see what the problem is?

oliversturm, 2020-07-25:
One more finding: if I start rigctld manually from the command line, things suddenly work in cqrlog. I can see my frequency and other status, and I can control the rig as expected. 

However, on the command line I see messages from rigctld that don't look quite right:

```
/usr/bin/rigctld -m 1040 -r /dev/ttyUSB0 -t 4532 -s 38400 -vvv
Recommend using --vfo switch for rigctld if client supports it
rigctl and netrigctl will automatically detect vfo mode
newcat_get_cmd: Rig busy
newcat_get_cmd: Rig busy
Opened rig model 1040, 'FT-DX101D'
newcat_get_cmd: Rig busy
newcat_get_cmd: Rig busy
newcat_set_cmd: Rig busy - retrying
newcat_set_cmd: Rig busy - retrying
newcat_set_cmd: Rig busy - retrying
```

I guess this is just a curiosity - and perhaps I should try to get hold of the very latest version of Hamlib, since I understand the FT-DX101 drivers are quite new.

But the fact remains that the original problem goes away if I run rigctld manually - even though I'm using the exact same command line.

OH1KH, 2020-07-26:
First of all, drop the poll rate. See what happens with 2000 (2sec). Your qsos will last well over 2sec, so you will have the right frequency reading when it is time to save qso.

You may also adjust other extra parameters. Start rigctrld form command line with "-L" to see all parameters:
There are write_delay, Post_write_delay, timeout and retry that might affect.

Open cqrlog with uncheck "start rigctld at program start"and set rig model 2 (Net Hamlib rigctld) , then close cqrlog.
To see what really happens start rigctld from command line with "/usr/bin/rigctld -m 1040 -r /dev/ttyUSB0 -t 4532 -s 38400 -vvvvv"
Leave it running. Be sure there is just one rigctld running, the one from console.
Now when starting cqrlog again it should connect to rigctld running at console and -vvvvv should give you a view  at console 
 what happens on rigctld side when cqrlog sends "fmv".
You can also use sudo tcpdump -X -ilo port 4532 to monitor the traffic between cqrlog and rigctld.

With latest Hamlib there has been reported problems with Yaesus, as well as Icoms but most of them should be fixed if you clone latest Hamlib from GitHub and compile it for use.

The way telnet sends "fmv" may differ from cqrlog's sending (compare TCP packets with tcpdump)

Let us know what happens.

OH1KH, 2020-07-26:
FYI: 
 (from tcpdump)

Telnet sends "fmv" like:
0x0030:  70b9 84ec 666d 760d 0a                   p...fmv..
and cqrlog sends:
0x0030:  70c8 0977 666d 760a                      p..wfmv.

So telnet adds 0x0d (CR) while cqrlog sends just =0x0a (LF) that is the difference.

oliversturm, 2020-07-27:
Thanks for all those ideas, certainly useful information. I'm afraid my initial messages may have made things sound more complicated than they appear now.

At this time, my setup is working correctly. I start rigctld manually (still using the command line I mentioned before). Then I start cqrlog and other applications as needed, and they can all control the rig and retrieve status information from the rig. I still see occasional warnings on the command line, but these don't seem to affect things - I'll see whether they go away with new Hamlib versions in the future.

I will change the title of this issue in a moment, because the issue report is now this: when cqrlog is configured to run rigctld on startup, it is unable to talk to the rig. If I run rigctld manually, with the same command line parameters, before starting cqrlog, everything works correctly.

mdblack98, 2020-07-27:
Make sure you put spaces between the commands "f m t" 
Otherwise it's trying to interpret it as one command.

Mike W9MDB

oliversturm, 2020-07-27:
Well, the cqrlog output says it's sending `fmt` with no spaces. I can do the same thing interactively with rigctl and it works correctly.

mdblack98, 2020-07-28:
My mistake...I'd never tried stacking the commands like that and didn't even know that worked....I'm used to stacking commands from the command line where you can't bunch them together like that.

There is a more recent version for the 101D -- perhaps you can compile it yourself?

Mike

mdblack98, 2020-07-29:
Another thing...ensure you don't have an old version of hamlib....like rigctl in /usr/bin
I imagine cqrlog will use the first one in the path unless you can specify the path.