ok2cqr/cqrlog #312: Populate ADIF field SAT_MODE

phl0, 2020-07-31:
During uploading logs for the AMSAT UK SAT QSO party I discovered that cqrlog does not populate SAT_MODE in ADIF exports. Would be nice to have this. Need to analyze which string is expected as content though.

See https://twitter.com/flo_0_/status/1289094605846257665

phl0, 2020-07-31:
ADIF spec is here:

https://www.adif.org/310/ADIF_310.htm#QSO_Field_SAT_MODE

But no sign of what is to be filled in. Maybe @magicbug can give a hint?

dl7oap, 2020-07-31:
look here at the end of the webside https://www.eqsl.cc/qslcard/ADIFContentSpecs.cfm

its the mode of the satellte like UV, VU etc.

so this would need a new database field and a new field in the qso to enter the satmode.

phl0, 2020-07-31:
Yepp. I just vreated this issue to remember after my return. For some satellites it may be hard-coded (AO-91) but others depend on modes (AO-92).
Would be good to have that in the database as well.

phl0, 2020-08-10:
Did a first commit into https://github.com/phl0/cqrlog/tree/SAT_mode
I copied most of the code from the getPropMode stuff. UI looks as follows:

[image]

phl0, 2020-08-11:
I have added some more code with 465f55903f1a6d05729c6327f1adf1a54cc36953 so that the database is updated with a new column and SAT mode info is stored into the database as well as read back on editing QSOs.
ToDos:
- (Pre-)set SAT mode after selection of a satellite (e.g. if SO-50 is selected make VU the preselected SAT mode)
- Update / set SAT mode depending on (TX) Frequency (and RX Frequency) (e.g. AO-92 can be UV or LV - so set to LV if TX frequency is 1267MHz or set to UV if TX frequency is 435MHz
- Include SAT_MODE field in ADIF exports
- Import SAT_MODE field from ADIF imports

phl0, 2020-08-12:
Export of SAT_MODE in ADIF/HTML was implemented with bb8aecde8ad88f87c6b53fa4ea1e0a622eb5c6f6.
So remaining tasks are:

- (Pre-)set SAT mode after selection of a satellite (e.g. if SO-50 is selected make VU the preselected SAT mode)
- Update / set SAT mode depending on (TX) Frequency (and RX Frequency) (e.g. AO-92 can be UV or LV - so set to LV if TX frequency is 1267MHz or set to UV if TX frequency is 435MHz
- Import SAT_MODE field from ADIF imports
- Update help files (with screenshots)

phl0, 2020-08-12:
Import of SAT_MODE from ADIF files has been committed with 95a1aae.
ToDo:

- (Pre-)set SAT mode after selection of a satellite (e.g. if SO-50 is selected make VU the preselected SAT mode)
- Update / set SAT mode depending on (TX) Frequency (and RX Frequency) (e.g. AO-92 can be UV or LV - so set to LV if TX frequency is 1267MHz or set to UV if TX frequency is 435MHz
- Update help files (with screenshots)

phl0, 2020-08-17:
The (Pre-)set SAT mode has been implemented with commit 4821c1f2552719130df3584daa59cedfbc6caf31.
Open issues:

- Update / set SAT mode depending on (TX) Frequency (and RX Frequency) (e.g. AO-92 can be UV or LV - so set to LV if TX frequency is 1267MHz or set to UV if TX frequency is 435MHz
- Update help files (with screenshots)

phl0, 2020-08-17:
Also SAT_MODE has also been modified to include a forward between the up- and downlink designators.

[image]

phl0, 2020-08-24:
As @dl7oap noted correctly this data is redundant in the cqrlog database because it can be derived from freq and rxfreq at any time. In parallel I noticed that eqsl.cc is obviously importing this field from ADIF uploads. So I made a new branch which can be found here:
https://github.com/phl0/cqrlog/tree/eQSL_SAT_mode
With this code the SAT_MODE field is computed on the fly (if enabled in export preferences) while exporting to ADIF/HTML and also inserted for eqsl.cc uploads.

phl0, 2020-08-30:
Cleaned up the code a made a separate new PR for this:
https://github.com/ok2cqr/cqrlog/pull/322
Will close here as soon as this is merged into master.

phl0, 2020-09-02:
PR https://github.com/ok2cqr/cqrlog/pull/322 has been merged. So closing here.

closed 2020-09-02 by phl0