ok2cqr/cqrlog #364: Pressing ESC while in QSO delete dialog deletes the QSO(s)

phl0, 2020-12-14:
There seems to be an issue in the dialog box that asks for confirmation of deletion of QSOs.
If I press ESC the selected (QSO)s does get deleted which I expect not to be the case.
Pressing ESC should cancel the dialog in order to not accidentially delete QSOs. Maybe "NO" should be the default selected answer button here.

phl0, 2020-12-14:
Hmm strange even with NO as default selected button pressing ESC leads to deleting the QSO(s) ....

'''
diff --git a/src/fMain.pas b/src/fMain.pas
index c9b4b7d1..8d064667 100644
--- a/src/fMain.pas
+++ b/src/fMain.pas
@@ -735,7 +735,7 @@ begin
     if dbgrdMain.SelectedRows.Count < 1 then
     begin
       if Application.MessageBox('Do you really want to delete this QSO?',
-        'Question ...', MB_ICONQUESTION + MB_YESNO) = idNo then
+        'Question ...', MB_ICONQUESTION + MB_YESNO + MB_DEFBUTTON2) = idNo then
         exit;
       dmData.qCQRLOG.DisableControls;
       try
@@ -761,7 +761,7 @@ begin
     else
     begin
       if Application.MessageBox('Do you really want to delete selected QSOs?',
-        'Question ...', MB_ICONQUESTION + MB_YESNO) = idNo then
+        'Question ...', MB_ICONQUESTION + MB_YESNO + MB_DEFBUTTON2) = idNo then
         exit;
       dmData.qCQRLOG.DisableControls;
       try
'''

OH1KH, 2020-12-14:
Confirmed !

Seems to be common problem with Lazarus.

Select "Callbook/Database update" it asks "Do you really want to run database update". If ESC pressed it is counted as YES.
Same in QSL label printing. Do not set filter, then it asks "do you..." Pressing ESC continues.

I guess also all "do you really want..." questions results same with ESC.

OH1KH, 2020-12-14:
See:
https://forum.lazarus.freepascal.org/index.php?topic=25614.0

OH1KH, 2020-12-14:
There are 182 Application.MessageBoxes in source to check...   ;-(  Not all needing fix, but some do.

phl0, 2020-12-14:
I have a proposed fix here:
https://github.com/phl0/cqrlog/commit/972a81ed982c0231790853569bd96f9e9b32a17b
Will be PRred if you agree :)

phl0, 2020-12-15:
I created PR https://github.com/ok2cqr/cqrlog/pull/367 to address this issue. Code has been crosschecked by @OH1KH. Thank you.

phl0, 2020-12-16:
Has been fixed with merging of https://github.com/ok2cqr/cqrlog/pull/367. So closing here.
Thank you very much @ok2cqr.

closed 2020-12-16 by phl0