ok2cqr/cqrlog #393: Importing dBASE III QSLs

SCOTT-HAMILTON, 2021-02-04:
Hello, can cqrlog import dBASE III QSL files to its SQL database ?

I don't know much about this technology but I have 2 files : 
 > QSLADDR.IXT: dBase III DBT, version number 0, next free block index 65535
and
 > QSLADDR.DAT: FoxPro FPT, blocks size 47622, next free block index 4294901760, field type 0 dBase III DBT, version number 0, next free block index 65535

Those can still be opened with an old MS-DOS program called PROPAGS which I could make to run with dosbox but it would be so much better to have them on a more modern file format compatible with such a software like cqrlog.

OH1KH, 2021-02-05:
HI !
I do not now direct answer but with one minute Googling I found:
https://stackoverflow.com/questions/14270236/php-script-to-convert-dbf-files-to-mysql

That could be good start for process.

Actually you just need to convert DBaseIII to text and produce adif tags from every qso and then import adif to cqlog (or what ever logging program). I think that way could be easier that trying to create mysql database tables.

Can your PROGPAS do text dump from all contents of your database? After that is just a play with AWK, perl, python or other tools to make text to adif.

SCOTT-HAMILTON, 2021-04-25:
Ok, thanks !

closed 2021-04-25 by SCOTT-HAMILTON