ok2cqr/cqrlog #411: Make cqrlog available as a Flatpak

chrisdebian, 2021-03-18:
Hi, i’m a GNU/Linux user, and have been hopping between distros, to find the best integration of cqrlog, with other functionality.

‌

Some distros have very old versions of the application [0], where as others have the current version, but with dependency issues. I wondered, has consideration been given to creating a Flatpak [1] version of the app?

‌

[0] https://repology.org/project/cqrlog/versions

‌

[1] https://flathub.org/home

‌

The advantage of using Flatpak is that all dependencies are managed, and unlike Snap, it is LGPL, where as Snap uses a proprietary backend.

‌

This would make adoption by Linux users, much easier, and would help ensure wider adoption.

‌

Thanks,

‌

Chris

2E0FRU

IO81vk

nolith, 2021-03-26:
This will be great! It could also help to have an up-to-date version for raspberrypi.

OH1KH, 2021-03-26:
Did you read issue #110 ?
Petr has tried several different ways to do complete package, appimage was latest.

All those have problems. Some with mysql database in user's folder, some with serial port connections, etc.

IMHO is that easiest way is to install *any* cqrlog version from package (that install add dependencies)  then upgrade it by making a compile from latest source. I have made few messages to cqrlog forum how to do it. Latest was for Raspberry few days ago.
Really, really easy.

nolith, 2021-03-26:
Thanks for the info @OH1KH 

I've never used flatpak before but I see that ftldigi and ftlrig are supported. So I guess serial port connection is not an issue. Maybe embedding mysql is not possible also there.

My experience with compiling cgrlog in the rpi3 was an unresponsive system due to OOM and filling entirely the default swap image, as I result I had a rpi with constant disk access and ssh connections refreshing screen no more than  2 times each hours.

Today I incremented the swap space and compile again successfully.

Given the rpi is quite a popular ham device, will it be possible to provide deb packages for the last Pi OS (debian Buster 32-bit at the moment) instead?

sp6ina, 2023-08-03:
Any progress ??

OH1KH, 2023-08-04:
HI!
I have compiled my Alpha binary version for Buster 10 armv7l (32bit) done with RPi4b 4Gb, but no install package of any kind. I'm not familiar with any package methods.
Just an update script that can do version swapping and backups (in my GitHub).

Maybe SomeOneElse(tm) does it?

ok2cqr, 2026-10-08:
CQRLOG already has flatpak available in the Releases section.

closed 2026-10-08 by ok2cqr