ok2cqr/cqrlog #433: Details window: ITU, WAZ confirmed

OH1KH, 2021-08-25:
Details window uses **_only_** qsl_r to check if are ITU & WAZ  confirmed.

Should it use at least LoTW in addition?
Is there a rule how qsos must be confirmed in these cases???  Card?, eQSl?, LoTW?
Now if paper qsls are not used details always claims that ITU and WAZ needs confirmation.

Fix is just a modification of sql sentence.

Comments, please.

phl0, 2021-08-25:
I think this is a good solution. Why not use any of the QSL recevied infos to display worked zones? So regardless whether it was paper, LotW or eqsl.cc ....

OH1KH, 2021-08-25:
HI Florian!
any or all ?
I meant that confirmed ITU or WAZ would be created using "where (qsl_r='Q' **or lotw_qsl_rcvd='L' or eqsl_qsl_rcvd='E'**)" but I am not sure are LoTW and  eQSL officially accepted ways to confirm  ITU and WAZ qsos.

QSL is always ok, but LoTW and/or eQSL are not accepted for all certificates and I do not know what is the situation with ITU and WAZ. Have to Google...

OH1KH, 2021-08-25:
OK!
I could have done this right away, but liked to start some conversation :-)
CQ says:
Electronic verifications from sources approved by CQ are acceptable. See the CQ website for acceptable online sources.
 
But to find acceptable online source needs work. Any of above sentence's words do not give any hits when searched with CQ's search. Suddenly they are found under "On-Line Confirmations" where they talk about " electronic confirmations" instead of "Electronic verifications" that is used in first sentence.

Sherlok Holmes is needed..... But WAZ seems to be OK with both LoTW and eQSL.

Then RSGB. It was easier to find:
Confirmation of two-way contacts is required by QSL cards, LoTW or Clublog with any amateur radio station within the ITU Zones listed at the time of application.

So the eQSL is not valid for ITU, but LoTW is.

It seems that with these rules I can make changes to details window.
I am afraid that this has snow ball effect also to statistics. I have not checked yet but I hope there is procedure somewhere in utils that is used for every places.

closed 2021-11-15 by OH1KH