ok2cqr/cqrlog #438: Cqrlog's future with HamLib

OH1KH, 2021-09-10:
Hi!
With the very latest Hamlib and wsjt-x there comes a situation that wsjt-x needs rigctld's start parameter "--vfo" to work properly with split/rig mode.
How ever this starting parameter changes all one letter rigctld commands to need used vfo as first parameter:
F 3500000 changes to F VFOA 3500000 (if change is wanted to A vfo) and so on.

I have had long conversations with Mike, W9MBD, and he has told that maybe in ver 5.0 the "--vfo" parameter is built in and can not be changed.
He also claims: "CQRLog chose to not use the Hamlib API and used the undocumented rigctld protocol." 
(!?!?!?) As far as I can see the TCP one letter commands are documented in rigctld man page!

I made a look to cqrlog's TRXcontrol/rigControl code and it looks more or less hmmm... unclear and confusing.

When it comes a time to do something it might be easier to rewrite all again.
But if Hamlib API is used I think we are then tied to compile Hamlib always together with cqrlog. I have not studied Hamlib API documentation yet. (where it is?) 

As an alternative we could perhaps use flrig instead.
Flrig has rig control, memories and cw keying built in. Unfortunately CW keying uses DTR/RTS and not rig's CAT. (at least I have not succeeded to get it work that way) so we are then loosing rig CAT based keyer like Hamlib had.

On the other hand we could then drop away TRXControl and rigContol because flrig has them. We would just need connection to flrig (TCP/XML) to get rig's frequency and mode. And also set them in case of jumping to DX spot frequency. Everything else for rig is handled by flrig.

This sounds quite easy.

Any opinions?

P.S.
While there still is "--vfo" startup parameter current problem can have workaround so that rigctld is started before cqrlog with "--vfo" and rig's parameters  and then cqrlog starts another rigctld with -m2 -t 14532 (without "--vfo" prameter) and polls it, where it chains poll to first rigctld.
It seems to work, but in current version the Hamlib keying causes error then. But I think root reason for that is that keying does not work anyway when "--vfo" parameter is used (bug somewhere, should not be vfo related).

lx2gt, 2021-09-10:
Hello,

In the past, cqrlog used the API, and was recommended to use rigctl directly by the hamlib people back then:

https://sourceforge.net/p/hamlib/mailman/message/21359880/

Having to use flrig does add complexity for the end user, I for one basically have never TRXcontrol window open if I don't have to, and am just happy to automatically get frequency and mode for my logged contacts.

Using the API might on the other hand be difficult... are there existing Free Pascal bindings?

Luc, LX2GT

OH1KH, 2021-09-13:
Thanks Luc!
Interesting old message.

Getting just the frequency/mode/vfo needs some fixes if parameter "--vfo" is used. Unfortunately even Hamlib version have differences how to poll then.
I think I now have version that can check the existence of "--vfo" startup parameter and can format the regular polling string "fmv" to suitable formats for Hamlib v4.3 and v3.3
Still the other commands need fixing but as I go along I have noticed already few bugs at Hamlib side.

I also tried flrig interfacing but got unexpected problems with xml encode and decode. I still have temptation to try also that because flrig gives so lot more versatile panel than TRXControl for users who want to drive their rigs via computer (perhaps even remote). 

I try to finish the Hamlib side first. And then think more about flrig.