ok2cqr/cqrlog #442: Incorrect value for 10GHz frequency

sq6emm, 2021-09-24:
Hi,

on cqrlog 2.5.2 when you log QSO on 10GHz band you will see frequency like:

1,0250.000 instead of 10.250.000 you can see this on QSO list and when entering QSO.

vy 73 de SQ6EMM

phl0, 2021-09-25:
Having a quick look at the sources I couldn't identify the error. But I have an idea. Maybe the frequencies > 10GHz are not formatted properly. 
Could you please check if this also applies to the default settings for 24, 47 and 78 Ghz pse?

sq6emm, 2021-09-25:
Hi,

yes it looks like formatting problem, the same on all high bands.

OH1KH, 2021-09-26:
Could it be how system locale is set to format numbers?

sq6emm, 2021-09-26:
Good question..
I am using Ubuntu 20.04 LTS (with all updates)
Below locale:

```
LANG=en_US.UTF-8
LANGUAGE=
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC=pl_PL.UTF-8
LC_TIME=pl_PL.UTF-8
LC_COLLATE="en_US.UTF-8"
LC_MONETARY=pl_PL.UTF-8
LC_MESSAGES="en_US.UTF-8"
LC_PAPER=pl_PL.UTF-8
LC_NAME=pl_PL.UTF-8
LC_ADDRESS=pl_PL.UTF-8
LC_TELEPHONE=pl_PL.UTF-8
LC_MEASUREMENT=pl_PL.UTF-8
LC_IDENTIFICATION=pl_PL.UTF-8
LC_ALL=
```

For test I've run:

`LC_ALL=en_US.utf8 cqrlog`

Issue remains the same.

sq6emm, 2021-09-26:
Line 146 of https://github.com/ok2cqr/cqrlog/blob/master/src/fFreq.pas

Looks suspicious to me.

sq6emm, 2021-09-26:
Fixed in: https://github.com/ok2cqr/cqrlog/pull/443

sq6emm, 2021-09-26:
Done... Please review;)

phl0, 2021-09-30:
I can confirm that this fix works.:

[image]

All GHz Band get logged correctly now. Thank you for fixing this. Guess I need to build some equipment for those bands now :-)

sq6emm, 2021-09-30:
Yea... would be great to test hamlib + LO shift with this;)

phl0, 2021-09-30:
tbh I don't know how that works. The bands table allows for configuration of RX/TX offsets. But I guess I need to read the manual about how these work.

sq6emm, 2021-10-05:
Changes, merged in #443

closed 2021-10-05 by sq6emm