ok2cqr/cqrlog #465: GTK2 Combobox Issue with Adapta-Nokto Theme

pulponair, 2021-11-25:
Problem:
Combo Boxes are not rendered correctly.
Not sure if that is a theme issue or a cqrlog issue. The QT version works ok.

If it is a theme issue i would be happy to report to the theme Author. Please advice :)

Version:
Master latest as of today (commit 30264390d909ee106944a65a534f5e7cf4a7c623)

OS
Linux mint Cinamon Linux Mint 20.2 Uma  

Screenshots 
[image]

OH1KH, 2021-11-26:
HI!
GTK and QT5 have always differences, but usually GTK has worked "out of box" and making QT5 similar has needed fine tuning with layout and anchoring.
That may be theme issue. Could you test with some other theme(s)?
With my Fedora 34/Openbox LXDE/Adwaita theme everything looks normal and equal with GTK and QT5. Image is with GTK.

BTW, where is the upper part and bottom status line of your NewQSO window?

[image]

pulponair, 2021-11-26:
@OH1KH 
Yepp you are right that seems to be a theme issue. Obviously the GTK2 setting of that theme are somehow broken for Comboxes. I tried to fix it myself but i am not very into GTK theming. I opened an issue over here.

https://github.com/linuxmint/cinnamon-spices-themes/issues/720

You might want to leave this issue open until they responded and may be to help others running into this.

Thank you for your work.

closed 2026-10-08 by ok2cqr