ok2cqr/cqrlog #476: Problem upgrading from 2.4 to 2.5.2

FabC, 2022-01-05:
I was not able to upgrade from CQRLOG 2.4.0-3 to version 2.5.2 with

**sudo add-apt-repository ppa:ok2cqr/ppa; sudo apt-get update; sudo apt-get install cqrlog**

because of the following error (in italian, sorry):

dpkg: errore nell'elaborare l'archivio /var/cache/apt/archives/cqrlog_2.5.2-1~focal_amd64.deb (--unpack): **tentata sovrascrittura di "/usr/share/cqrlog/changelog.html" presente anche nel pacchetto cqrlog-data 2.4.0-3dpkg-deb**: errore: il sottoprocesso paste è stato terminato dal segnale (Pipe interrotta)
Changes removed from /etc/apparmor.d/local/usr.sbin.mysqldRestarting apparmor...

Changes removed from /etc/apparmor.d/usr.sbin.mysqld

File /etc/apparmor.d/local/usr.sbin.mysqld was modified, restarting apparmor

Si sono verificati degli errori nell'elaborazione: /var/cache/apt/archives/cqrlog_2.5.2-1~focal_amd64.debE: Sub-process /usr/bin/dpkg returned an error code (1)

The upgrade process fails because it tries to it tries to overwrite the changelog file.

Working on Linux Mint 20.2

Thanks for the support,
73, Fabrizio IU5GEZ

OH1KH, 2022-01-05:
HI!
 This problem is discussed some time ago at https://www.cqrlog.com/forum/
 
Before upgrading cqrlog you must remove package **cqrlog-data 2.4.0-3dpkg-deb** it is deb package maintainer's innovation and is not needed with cqrlog. The ppa you have added contains same data in cqrlog package. (actually question is contents of /usr/share/cqrlog folder in this strange data-package, I think)

FabC, 2022-01-05:
Perfect, **sudo apt-get remove cqrlog-data** solved the problem.
Thanks
73, Fabrizio IU5GEZ

closed 2022-01-05 by FabC