ok2cqr/cqrlog #514: BandMap - cannot set frequency for some items

peterdulowsky, 2022-06-06:
https://github.com/ok2cqr/cqrlog/blob/823332ccb2e56de59f0f03ddee8834fbeb5b5c9c/src/fBandMap.pas#L315

This function returns an invalid (non existent item) and therefore the frequency nor callsign is set in new QSO window. For most items it works, for some it does not. 

E.g. Clicke to 6, array pos 46 in bandmap with 8 items. All other items worked fine. See log below.

Captured log:
[CRITICAL] clearlooks_style_draw_focus: assertion 'width >= -1' failed
Clicked to:6
GetIndexFromPosition, looking for:   14042.000   GB70E 
Array pos: 46
SELECT * FROM profiles WHERE visible > 0 ORDER BY nr

SELECT * FROM profiles WHERE nr = 1

SELECT * FROM cqrlog_common.bands WHERE band = ''

select rx_offset, tx_offset from cqrlog_common.bands where b_begin <= :b_begin and b_end >= :b_end

BandMapItems[1].Freq: 2.8074000000170003E+004
BandMapItems[1].Call:G8DDR
BandMapItems[1].Position:200
BandMapItems[2].Freq: 2.8074000000098000E+004
BandMapItems[2].Call:GQ0GFC
BandMapItems[2].Position:200
BandMapItems[3].Freq: 2.8074000000091997E+004
BandMapItems[3].Call:G4CJC
BandMapItems[3].Position:200
BandMapItems[4].Freq: 2.8074000000077001E+004
BandMapItems[4].Call:M3OPG
BandMapItems[4].Position:199
BandMapItems[5].Freq: 1.8082500000167001E+004
BandMapItems[5].Call:GB70M
BandMapItems[5].Position:197
BandMapItems[6].Freq: 1.8072000000144002E+004
BandMapItems[6].Call:GQ3SVK
BandMapItems[6].Position:198
BandMapItems[7].Freq: 1.4042000000126000E+004
BandMapItems[7].Call:GB70E
BandMapItems[7].Position:199
BandMapItems[8].Freq: 7.0460000001119997E+003
BandMapItems[8].Call:GB70E
BandMapItems[8].Position:198
Sending: +F 0

Sending: +M  500

Msg from rig: SET_FREQ: 0
RPRT -9

OH1KH, 2022-06-07:
To track this it would be better to add few writeln lines.
Specially interested what happens for frequency  at line 302
procedure TfrmBandMap.EmitBandMapClick(Sender:TObject;Call,Mode : String; Freq : Currency);

then fNewQSO.pas line 7221 when entering
procedure TfrmNewQSO.OnBandMapClick(Sender:TObject;Call,Mode: String;Freq:Currency);  

and what happens when entering fNewQSO.pas  line 6676
procedure TfrmNewQSO.NewQSOFromSpot(call,freq,mode : String;FromRbn : Boolean = False);

Is the frequency still valid over all procedures.

It is better to disable debug and add only those writelns to print the frequency. That way it is easier to follow.
I have never noticed that kid of bug, but I do not use bandmap a lot and always with "show only current band"

peterdulowsky, 2022-06-07:
My gut feeling tells me that the procedure GetIndexFromPosition returns wrongly calculated Array pos, i.e. Array pos: 46 in an array that consists of 8 items. And the following code is trying to access a non existent item in BandMapItems array. 

This is a log for a correctly set freq after doublicliking to BandMap item (same content of bandmap as above):

[CRITICAL] clearlooks_style_draw_focus: assertion 'width >= -1' failed
**Clicked to:5
GetIndexFromPosition, looking for:   18072.000  GQ3SVK 
Array pos: 6**
SELECT * FROM profiles WHERE visible > 0 ORDER BY nr

SELECT * FROM profiles WHERE nr = 1

SELECT * FROM cqrlog_common.bands WHERE band = '17M'

select rx_offset, tx_offset from cqrlog_common.bands where b_begin <= :b_begin and b_end >= :b_end

Found - GQ3SVK
SELECT * FROM view_cqrlog_main_by_qsodate WHERE idcall = 'GQ3SVK' ORDER BY qsodate,time_on

LoadForm: frmNewQSO
SELECT id_cqrlog_main FROM cqrlog_main WHERE adif=223 AND band='17M' AND ((qsl_r='Q') OR (lotw_qslr='L')) AND mode='CW' LIMIT 1

SELECT id_cqrlog_main FROM cqrlog_main WHERE adif=223 AND band='17M' AND mode='CW' LIMIT 1

SELECT id_cqrlog_main FROM cqrlog_main WHERE adif=223 AND band='17M' LIMIT 1

select band,mode,qsl_r,lotw_qslr,eqsl_qsl_rcvd from cqrlog_main where adif=223 and ((qsl_r='Q') or (lotw_qslr = 'L') or (eqsl_qsl_rcvd='E')) group by band,mode,qsl_r,lotw_qslr,eqsl_qsl_rcvd

select band,mode from cqrlog_main where adif=223 group by band,mode

-------------------------------------------------------------------

OH1KH, 2022-06-07:
Yep!
I did not look carefully enough. This bandmap is not very easy to follow.
There is some kind of thread. Could it be that delete from bandmap is processed while the GetIndex is callled from other thread.
At suitable moment there could be callsign index found, but after a nanosecond that index is cleared giving F=0 That is just a guess, I did not follow code carefully enough to see is that even possible.

That could explain the random/seldom fail. That kind of errors are very hard to catch.

peterdulowsky, 2022-06-08:
I don't think that index has been cleared in the background. The content of the bandmap has not been changed between the tests above. I tend to believe there might be a connection between to the font size and perhaps the size of the bandmap window.  After changing the font size to 8 from 10 (better suits the screen size and window arrangement) I was not able to reproduce the issue.
I will try to change it back to 10 to see if it makes any difference.
For reference I am running cqrlog on RPi3 with stock raspbian OS.