ok2cqr/cqrlog #531: Overlapping fonts on HiDPI display

chrism0dwk, 2022-07-20:
Hi There,

Firstly, thanks for what looks like an awesome piece of software!  I've been QRT since 2012 and just getting back into the hobby.  I'm a 100% Linux user, and scientific programmer, and I'm amazed at the work that's been done in the Linux amateur radio space over the last 10 years!

I'm running Mint 20.3 (upstream Ubuntu Focal) with the Cinnamon desktop, CQRLog installed from the PPA as indicated in the install instructions (v. 2.5.2).  

My display uses HiDPI resolution (3840x2160) driven by NVIDIA proprietary drivers.  Unfortunately, I'm seeing really misaligned fonts as in the attached screenshot.  If you could zero me in on a particular bit of the source code, I could have a play and see if I can improve matters.

Any thoughts gratefully received.

73

Chris, M0DWK
[image]

OH1KH, 2022-07-21:
Hi Chris!

I am afraid that is the Lazarus/FPC problem with it's GTK2 widget code.
You could try to compile Cqrlog with QT5 widgets and see what kind of problems that gives with HiDPI. I am sure it looks different, but how (I do not have HiDPI monitors).

See this message https://www.cqrlog.com/node/2998 

You have to install "sudo apt install libqt5pas1 libqt5pas-dev" and then compile from command console with "make cqrlog_qt5" or from Lazarus-ide by setting Project/Project options/Additions and overrides/LCL widget type and after that Run/compile again.

[image]

This is more Lazarus/Freepascal problem than Cqrlog's, I think.

chrism0dwk, 2022-07-21:
Thanks for the advice.  Interestingly, building from source with QT5 did not immediately sort out the issue.
[image]

However, resizing the fonts to 8pt did at least decrease the text size so that overlaps were avoided.   This is a solution that works in both cases, but leads to an uncomfortably small fontsize.  Usable, though!

73,

Chris

OH1KH, 2022-07-22:
HI!
Did you try to set scaling to 100%?
Seet the last message of this chain https://www.cqrlog.com/node/1454

lukolszewski, 2025-01-30:
Yes, you can set scaling to 100%, another equally good solution is to get rid of your HiDpi monitor because now that you removed scaling you can no longer read anything.....

lukolszewski, 2025-01-30:
Lazarus wiki has this article about HiDpi https://wiki.lazarus.freepascal.org/High_DPI Is this something we can use? I never uised Lazarus and my last contact with Pascal was 40 years ago (I was hoping it will stay that way, but I do want to use cqrlog and it is unacceptable how it works on hidpi displays so looks like I have to look at it). I just need some pointers where to look.

lukolszewski, 2025-01-30:
And the version built with this Lazarus IDE runs fine with my 1.8x font scaling. Did you fix it? Or is this some change in Lazarus?

[image]

OH1KH, 2025-01-31:
Hi!
Have you tried this?

[image]

I can not see any difference with or without it, or test it in any other way as I do not have any HI-DPI monitor.

Link you referred has also some hints for Linux / Xorg but now Wayland is pushing in and it is totally different graphical environment mixing up everything.

MeToo.... with Pascal. Started in 1982 and there has been shorter and longer caps with programming. But living was then easier with 80x25 char Green CRTs and 300baud Teletype terminals.