ok2cqr/cqrlog #534: No dark mode on any window

vitodoc, 2022-10-07:
Dark mode hasn't worked for several months.
It is impossible to use the program for a long time with a white background, it hurts the eyes.
Please fix

OH1KH, 2022-10-08:
"Dark mode" is not Cqrlog feature. It comes from you Xwindow system. And needs QT5
I assume something has happened to your system (possible update(s) has changed something).

Here is the view from Cqrlog 2.6.0(001) compiled from current source code with QT5 widgets today 2020-10-08
[image]

blackbicbiro, 2024-03-19:
dark theme on Pop_os! works when OS is set to dark theme:-)

closed 2026-10-08 by ok2cqr