ok2cqr/cqrlog #536: Can not set frequency for FT-897D from CQRLOG (readout works)

Yume303, 2022-10-23:
Can not set frequency/mode from CQRLOG for my rig (readout works nicely).
Also tested with flrig and wsjtx with the same settings and it works (also tested under windows with flrig, and it works there too).
rigctld/hamlib version: 4.3.1

OH1KH, 2022-10-24:
There is difference with wsjtx and flrig. Wsjtx uses Hamlib API and flrig uses it's own procedure (if not configured to use Hamlib, when it uses Hamlib API).

Cqrlog uses rigctld string commands that you can simulate making telnet connection to localhost:4532
Set preferences/TRXConsole/Show communication with TRX in console checkbox checked. Close Cqrlog and start it from command console typing:
cqrlog

When you see what Cqrlog sends and how rig answers you can test it with telnet localhost 4532 manually if needed.
It may be wise to set poll rate to something like 2000 to get more time to see what happens.

Report the problem you see.

Yume303, 2022-10-24:
Thanks for the quick answer. Some additional information for the above: i tested flrig with hamlib (hamlib api as I see for now).
Also, executed a new test now with rigctld and gpredict, it works and sets frequency / mode correctly with the same cli arguments as the cqrlog's rigctld instance(except the -vvvv for debugging).

Messages from CQRLOG:
Settings:
-----------------------------------------------------
RigCtldPath:rigctld
RigCtldArgs:-m 1043 -r /dev/ttyUSB0 -t 4532  -s 38400  --set-conf=data_bits=8,stop_bits=1,serial_parity=None,rts_state=OFF
RunRigCtld: TRUE
RigDevice:  /dev/ttyUSB0
RigCtldPort:4532
RigCtldHost:localhost
RigPoll:    2000
RigSendCWR: FALSE
RigId:      1043

Starting RigCtld ...
rigProcess.Executable: rigctld
Parameters:
-m
1043
-r
/dev/ttyUSB0
-t
4532
-s
38400
--set-conf=data_bits=8,stop_bits=1,serial_parity=None,rts_state=OFF

rigctld started!
Connected to rigctld @ localhost:4532
Sending: � 1

Msg from rig: RPRT 0
Sending: fmv

Msg from rig: 145889340
Msg from rig: PKTUSB
0
Msg from rig: VFOA
Sending: fmv

Msg from rig: 145889340
PKTUSB
0
Msg from rig: VFOA
Sending: fmv

Msg from rig: 145889340
PKTUSB
0
Msg from rig: VFOA
Sending: fmv

Msg from rig: 145889340
Msg from rig: PKTUSB
0
Msg from rig: VFOA
Sending: fmv

From command line with gpredict:

yume@csotanyodu:~ $ rigctld -m 1043 -r /dev/ttyUSB0 -t 4532  -s 38400  --set-conf=data_bits=8,stop_bits=1,serial_parity=None,rts_state=OFF -vvvv
../../tests/rigctld.c(581) Startup: rigctld -m 1043 -r /dev/ttyUSB0 -t 4532 -s 38400 --set-conf=data_bits=8,stop_bits=1,serial_parity=None,rts_state=OFF -vvvv
rigctld Hamlib 4.3.1 Mon Sep 13 23:12:03 2021 +0000 SHA=c2ed3d
Report bugs to <hamlib-developer@lists.sourceforge.net>

yaesu: initrigs4_yaesu called
rig_register called
rig_register: rig_register (1021)
[... 397 lines cut ...]
rig.c(1954):rig_set_freq return(0)
rigctl_parse.c(1995):rigctl_set_freq return(0)
sync_callback: client lock disengaged
rigctl_parse.c(1758):rigctl_parse return(0)
rigctl_parse.c(651):rigctl_parse entered
sync_callback: client lock engaged
rigctl_parse.c(2007):rigctl_get_freq entered
rig_get_freq(1993) called vfo=currVFO
../../src/rig.c(2002) vfo=currVFO, curr_vfo=VFOA
rig.c(2063):rig_get_freq return(0)
rigctl_parse.c(2033):rigctl_get_freq return(0)
sync_callback: client lock disengaged
rigctl_parse.c(1758):rigctl_parse return(0)
rigctl_parse.c(651):rigctl_parse entered
sync_callback: client lock engaged
rigctl_parse.c(2911):rigctl_set_split_vfo entered
rig_parse_vfo called
rigctl_parse.c(2927):rigctl_set_split_vfo return(-1)
sync_callback: client lock disengaged
rigctl_parse.c(1758):rigctl_parse return(-1)
handle_socket: rigctl_parse retcode=-1
rigctl_parse.c(651):rigctl_parse entered
rigctl_parse.c(801):rigctl_parse return(1)
handle_socket: rigctl_parse retcode=1
Connection closed from localhost:54610

Yume303, 2022-10-24:
Also cqrlog with debug (-vvvv) enabled:

`Settings:
-----------------------------------------------------
RigCtldPath:rigctld
RigCtldArgs:-m 1043 -r /dev/ttyUSB0 -t 4532 -vvvv -s 38400  --set-conf=data_bits=8,stop_bits=1,serial_parity=None,rts_state=OFF
RunRigCtld: TRUE
RigDevice:  /dev/ttyUSB0
RigCtldPort:4532
RigCtldHost:localhost
RigPoll:    2000
RigSendCWR: FALSE
RigId:      1043

Starting RigCtld ...
rigProcess.Executable: rigctld
Parameters:
-m
1043
-r
/dev/ttyUSB0
-t
4532
-vvvv
-s
38400
--set-conf=data_bits=8,stop_bits=1,serial_parity=None,rts_state=OFF

../../tests/rigctld.c(581) Startup: rigctld -m 1043 -r /dev/ttyUSB0 -t 4532 -vvvv -s 38400 --set-conf=data_bits=8,stop_bits=1,serial_parity=None,rts_state=OFF
rigctld Hamlib 4.3.1 Mon Sep 13 23:12:03 2021 +0000 SHA=c2ed3d
Report bugs to <hamlib-developer@lists.sourceforge.net>

yaesu: initrigs4_yaesu called
rig_register called
rig_register: rig_register (1021)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (1027)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (1010)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (1005)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (1006)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (1007)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (1039)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (1009)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (1020)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (1001)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (1022)
register.c(225):rig_register return(0)
rig_register called
rig_register: rig_register (1023)
register.c(225):rig_register return(0)
rig_register called
[... 291 lines cut ...]
rigctl_parse.c(2167):rigctl_get_mode return(0)
sync_callback: client lock disengaged
rigctl_parse.c(1758):rigctl_parse return(0)
rigctl_parse.c(651):rigctl_parse entered
sync_callback: client lock engaged
Msg from rig: PKTUSB
0
rigctl_parse.c(2218):rigctl_get_vfo entered
rig.c(2753):rig_get_vfo entered
ft857_get_vfo: called 
ft857_read_eeprom: called 
read_block called
rig.c(2809):rig_get_vfo return(0)
rigctl_parse.c(2234):rigctl_get_vfo return(0)
sync_callback: client lock disengaged
Msg from rig: VFOA
rigctl_parse.c(1758):rigctl_parse return(0)
rigctl_parse.c(651):rigctl_parse entered
1
1a
2
3
4
5
6`

OH1KH, 2022-10-25:
From Cqrlog part I can see it reads frequency and mode OK. As you said.

But let me see ONLY that part of debug where you try to SET frequency, if that is the defunct part.

What this have to do with gpredict?
Are you trying to run Cqrlog and Gpredict at same time?

That does not work unless you set rig model in Gpredict to be 2 (Hamlib net rigctld).
Two or more programs can not access same serial device without problems. Linux lets you do that (Windoze does not) but it does not work preperly.
Please read this https://github.com/OH1KH/cqrlog/blob/loc_testing/compiled/setting_rigctld_for_all_programs.pdf

Yume303, 2022-10-25:
Just tested it with gpredict to rule out hw/cable problem. Also tested one after one, not at the same time.
Trying to set from the freq from cqrlog does not "arrives" to rigctldas I see, nothing in the log (tried to grep through to find something like in the gpredict-test log, and checked manually too.)

OH1KH, 2022-10-25:
You say Cqrlog can not **set frequency**.

I am not looking for long list of polling (reading freq,mode, vfo). I like to see only the part where you click band button or  click frequency reading, type a new frequency and press enter  using TRXControl window. I.E. setting new frequency via Cqrlog.

It should look like this:
(NOTE: this is done with Cqrlog 2.6.0(001) using rig nr:1 Hamlib Dummy and looks a bit different than your older Cqrlog debug print)

Poll Sending: +f +m +v

Msg from rig:|GET_FREQ:
FREQUENCY: 145000000
RPRT 0|

Msg from rig:|GET_MODE:
MODE: FM
PASSBAND: 15000
RPRT 0|

Msg from rig:|GET_VFO:
VFO: VFOA
RPRT 0|
                                         < -----------------Pressed here 6m band button of TRXControl
**Queue Sending: [0] +F 51300000**

**Queue Sending: [1] +M FM 2500**

**Msg from rig:|SET_FREQ: 51300000**
**RPRT 0|**

**Msg from rig:|SET_MODE: FM 2500**
**RPRT 0|**

Poll Sending: +f +m +v

You should find similar dump when you try to **set** rig to another **frequency**.

Yume303, 2022-10-25:
That's where the problem is i think, tried again, only got these lines (tried to set from dropdown menu, entered freq manually, but at the next readout it reverted back on the gui to the original frequency(same as on the radio's lcd freq).
Maybe the set command doesn't make through the socket to rigctld (dunno), and because of that, it does not appears in the log?

Msg from rig: 145893080
Msg from rig: FM
0
Msg from rig: VFOB
Sending: fmv

Msg from rig: 145893080
FM
0
VFOB
Sending: fmv

Msg from rig: 145893080
Msg from rig: FM
0
Msg from rig: VFOB
Sending: fmv

Msg from rig: 145893080
Msg from rig: FM
0

OH1KH, 2022-10-27:
Ahhhh....
That is not bug. Now  I finally understood when you mentioned dropdown menu.

You can not drive rig from NewQSO window. It's purpose is just log the worked qso.
You should open NewQSO/Window/TRXControl window.  As the name says it is for controlling TRX.

Yume303, 2022-10-27:
Ohh... thanks for clarification, it wasn't clear for me, and above method just works fine.
Also sorry for wasting your time, regards and 73's !
Peter (HA1YME)

OH1KH, 2022-10-28:
No problem!
Main thing is that you got answer.

Sometimes, when time permits, look at the help section (NewQSO/Help/Help index) that open to your browser, please.
You find also some videos I have made from https://www.youtube.com/channel/UC3yPCVYmfeBzDSwTosOe2fQ

Please close this issue if you agree it is solved.
Thank you!

Yume303, 2022-10-28:
Can be closed, thanks a lot.

closed 2022-10-28 by Yume303