ok2cqr/cqrlog #555: libmariadbclient-dev dependency broken in recent debian/ubuntu releases

georgedorn, 2023-01-16:
`libmariadbclient-dev` has been removed, so releases can't be installed on anything that ships with mariadb more recent than 10.4.  It should be replaced with `libmariadb-dev` or `libmariadb-dev-compat` depending on what it actually needs.

georgedorn, 2023-01-16:
Similarly, `libhamlib2` is now `libhamlib4`.

georgedorn, 2023-01-16:
I can confirm that by replacing `libmariadbclient-dev` and `libmariadbclient-dev-compat` with `libmariadb-dev` and `libmariadb-dev-compat` in the `DEBIAN/control` file (along with swapping out `libhamlib2` for `libhamlib4`), I can install and run the latest release.

I don't quite know how to test whether libhamlib is still behaving as I'm still quite new to cqrlog, but I will be attempted to use it as my logging software in the very near future and will report back.

OH1KH, 2023-01-17:
Thanks for reporting!

I think there will be no problems with libhamlib as Cqrlog uses only rigctld and rotctld via telnet connections.
Of course the libhamlib must be same version as those two. Otherwise problems will arise.

That can happen if Hamlib is installed from package and uses /usr/bin for binary files and the user upgrades
Hamlib by self compile and uses /usr/local/bin without removing package installed Hamlib  (also some Ubuntus seems to do upgrade to /usr/local/bin leaving old binary to /usr/bin)
Then it might be that binary and library file versions get mixed while in use and it can make strange things (I have experience) 

IMHO I think that there should be a script that would install all dependencies for Cqrlog.
It could be fitted for different Linux brands so that it could find right names for packages.

That way we could get rid of 10^n version of older packages than the Cqrlog current source is, and there would be no need to wait for new ones to come.

After dependencies are installed from script then Cqrlog install is just a copy of few files (included in tgz file from download page).
Doing so also upgrade of dependency, like Hamlib, could be done without effect to Cqrlog. 
(Try to remove old Hamlib if it is installed with Cqrlog package, it removes also Cqrlog.)
Bad.

Once Cqrlog is installed upgrading it is just simply copying new /usr/bin/cqrlog and possible /usr/share/cqrlog/help contents. 
No need for packages with that.

NipponBill, 2023-09-24:
How can the .deb CQRLOG 2.5.2 be installed on Debian 11 or 12? This error prevents an install. 

 Is it recommended to move to the beta?

OH1KH, 2023-09-25:
That begins to be a real problem as packages of Cqrlog are so old compared to latest Linux releases that depended package list names do not fit to today's naming.

Most recommended way is to compile Cqrlog from source in same version of Linux than it is going to be used.
To manage with compile some tools should be installed. Most important is Lazarus that installs also lots of other stuff needed.
With Lazarus it is also best to check latest package from their web lazarus-ide.org as packages from Linux distribution can be far behind.

jbdubbs, 2023-11-13:
I solved this dependency issue by adding the MariaDB repo to Debian 12.

curl -LsS https://r.mariadb.com/downloads/mariadb_repo_setup | sudo bash

AsciiWolf, 2024-09-16:
Since version 2.5, CQRLOG [depends on mariadb](https://github.com/ok2cqr/cqrlog/issues/291#issuecomment-770383265) and does not work properly with original mysqld. There was no problem in Debian because it uses mariadb as its default mysql server, however it is broken in Ubuntu, (I believe) since 22.04, because Ubuntu uses mysqld as the default one.

It recently got [changed in Debian](https://salsa.debian.org/debian-hamradio-team/cqrlog/-/commit/33bad7fa4511f96d4ff48a3b868b583798ee92d6) to hard-depend on mariadb and a Stable Release Update for 24.04 (and 22.04) was prepared.

However, Ubuntu developers have rejected the update and are [not happy](https://bugs.launchpad.net/ubuntu/+source/cqrlog/+bug/1872002/comments/20) about it so it is possible that CQRLOG in Ubuntu (and its derivatives like Linux Mint) will sadly remain broken in future releases.

If you have any questions or comments, please, consider discussing them in the original Ubuntu ticket instead of this upstream one:

https://bugs.launchpad.net/ubuntu/+source/cqrlog/+bug/1872002

Thanks!

ok2cqr, 2026-10-12:
This should be already fixed.

closed 2026-10-12 by ok2cqr