ok2cqr/cqrlog #559: "Failed to open required defaults file" from PPA

cjritola, 2023-08-05:
Using PPA
Ubuntu 22.04.1
SSD usage 11%

```
mysqld: [ERROR] Failed to open required defaults file: /home/username/.config/cqrlog/database/mysql.cnf
mysqld: [ERROR] Fatal error in defaults handling. Program aborted!

```
Tried setting `mysql.cnf `permissions to 0644 but did not improve.
Tried 777 and it got further but that's not reasonable.
Tried adding mysql to my username's group but did not improve.

_Where I am now with this is after trying to fix an earlier problem:_

CQRLog from the ubuntu repos was running very slow. As per the[ troubleshooting page ](https://www.cqrlog.com/node/41)I attempted to set hamlib rig to dummy but CQRLog would freeze once OK is clicked in Preferences. With debug level 5, no output after clicking OK. Even Hamlib polling spam pauses.

`cqrlog.cfg` was blank.
Tried editing 1cqrlog.cfg with the hamlib mode 1 (dummy) instead of 2 (net). Copied 1cqrlog.cfg to the blank cqrlog.cfg. Started CQRLOG, opened Preferences - old settings persisted! Still in mode 2! Where the heck is it getting these settings?! Still freezes when clicking OK.

Tried `mv cqrlog.cfg cqrlog.bak.cfg` and `mv 1cqrlog.cfg 1cqrlog.bak.cfg` and no change. It's still loading the old settings from somewhere.

ran `watch ls -l *.cfg` after "OK" in prefs was clicked. `1cqrlog.cfg`'s size was randomly increasing and decreasing. It's as if it is writing the cfg file, deleting it, rewriting, in a permanent loop, in rapid succession.

Before declaring this as a bug I removed the CQRLOG package via APT and tried the PPA just in case it was fixed and broke it further.

OH1KH, 2023-08-07:
Check that you are using MariDB, not Mysql that is Ubuntu default database engine.

Configs are saved in database when Cqrlog is not running. While it is running you see Xcqrlog.cfg whre X is the open log's number. But you should not edit that file. Changes are not permanent and may not be taken account at all.

Right way is to open Cqrlog to first window "Database connection". There select log that you want configurations to be changed by manual edit. Do not open log.
Instead press "utils" and select configuration/export. Do you changes to that file with plain text editor and save again.
While still in "Database connection" window and have same log selected press again "utils" and select configuration/import and select the file you just edited.
After that you may open the log normal way,

If you create a new log you can use same procedure. Select new log, but do not open it. Instead do utils/configuration/import from previously saved file. The you can open new log and it has all the same settings than the log you used to make utils/configuration/export. Without any peek to preferences.

Settings are based to your log and will go with It's database, even when you move them with mysqldump: That's why the procedure is hard to find without knowing the idea.

cjritola, 2023-08-09:
Thanks, that help me get up and running again. I came across some oddities along the way:

Explicitly installing mariadb in APT was able to fix the issue- which removed regular mysql, but required cqrlog to be removed also. However, re-installing cqrlog afterward was possible and it didn't require any additional packages.

When exporting settings, a file can be saved without the .ini extension (i.e. `cqrlog.config`). However, when importing: only files ending in .ini are visible by the file selector.

I was able to change the TRX settings to dummy but am still having hanging issues. Will open a new issue for that.

closed 2023-08-09 by cjritola