ok2cqr/cqrlog #565: Autobackup Local Preference ignored

ik5pwc, 2023-10-22:
Use CQRLOG wih external database
Configured each computer to use local preferences for autobackup
On 1st PC autobackup is enabled locally
On 2nd PC autobackup is disabled locally

ISSUE: local autobackup configuration is not honored, always use the configured database value (i.e. enabling on PC1 will also enable on PC2 and viceversa, disabling on PC2 will also disable oin PC1)
[image]

OH1KH, 2023-10-24:
Thanks for reporting!

Found out that at least my Alpha 119 version has problems with bands, modes,rbn in addition with backup problem.
I did once learn how local/common was made and now had to do it again :( It is quite complicated to follow from source...

OH1KH, 2023-10-24:
Some study.
It seems that there are two missing things:
1) when configuration storage window is closed using "save" button lini (Xlocal.cfg TMemIniFile) must be flushed.
  Otherwise new settings  are appended to old contents .
2) When configuration storage window is closed using "save"  preferences must be closed by "OK" it may not be closed with cancel or top right corner "X" otherwise new contents of Xlocal.cfg file is not written.

- added lini file flush and "preferences close by cancel" disable to test this.
If lini is not flushed it is very difficult to get rid of old contents of Xlocal.cfg file because it is always "updated" and may cause feeling that some of local settings are not working.

Same problem exists with global configuration Xcqrlog.cfg file because it is also always "updated", not rewritten from zero, and it may contain old variable=value pairs that are not used any more. 
Because of that it is considered should preferences/program contain button that flushes both local and global cfg files. How ever it may be dangerous because not all settings are not loaded in preferences window when it opens.

Perhaps ini (Xcqrlog.cfg) flush could happen at Cqrlog close phase. It must be checked and tested.

OH1KH, 2023-10-24:
@ik5pwc 
Can you test local backup this way, please:
1) set preferences/Program/ConfigurationSettings/backup-Local computer only
2) close ConfigurationSettings from "save" button
3) close preferences from "OK" button.
4) open preferences/Exit & Auto backup and set path to save backup (use something different path like "/tmp")
5) close preferences from "OK" button.
6) close Cqrlog and answer yes if you have "Ask backup before close"
7) see that backup went to folder you selected.
8) open Cqrlog
9) set preferences/Program/ConfigurationSettings/backup-Common for all computers
10) close ConfigurationSettings from "save" button
11) close preferences from "OK" button.
12) open preferences/Exit & Auto backup and set path to save backup (use something different than with "local computer only")
13) close preferences from "OK" button.
14) close Cqrlog and answer yes if you have "Ask backup before close"
15) see that backup went to folder you now have selected.

If you do this exactly this order backup should work like expected. after phase 15) Xlocal.cfg file will still show section backup and path you entered at phase 4) but it is not taken account if  ConfigurationSettings/backup-Common for all computers is set. This is because Xlocal.cfg file is not properly flushed after change.

How ever without program fixes this order should work.

ik5pwc, 2023-10-24:
Hello @OH1KH , I don't exactly understand your test as my issue is not related to wrong path, but to enabling/disabling autobackup. BTW, in .config/cqrlog no Xlocal.cfg file is present, neither when cqrlog is running nor is closed.

Let me explain better.
PC1: Linux machine with mariadb instance running as daemon. CQRLog is configured to use a "remote" database (which in this case is hosted on the same computer). I want CQRLOG to perform autobackup on this computer
PC2: another linux machine where CQRLOG is configured to use PC1's mariadb. on PC2 I DON'T WANT autobackup.

I checked that local settings are properly saved on both computer within 1cqrlog.cfg (PC1 autobackup on, PC2 autobackup off) so it seems cqrlog properly saved my settings. However autobackup have the same behaviour on both computers (both ON or off, depending which was the latest setting).

I'll try to do the exact steps you mentioned but instead of changing the path I'll enable/disable the backup. Circle back in few minutes

ik5pwc, 2023-10-24:
Hello, did some more tests for this config and this is what I found (from experience)
1) CQRLOG saves in cqrlog_config table the list of "sections" to be managed locally 

```
[ConfigStorage]
Items=TRX,TRX1,TRX2,ROT,ROT1,ROT2,xplanet,CW,fldigi,wsjt,Backup,WindowSize,Window,Grayline,ColumnSize,
```
When cqrlog starts, it read all info from database and then try to open the local.cfg file reading the section that need to be "locally managed"

The important thing is to define **at the very beginning** which section manage locally and which manage using database. and confirm this setting before performing other (as you say in steps 1,2,3)

 Also discovered that probably there is a little error on value. When I looked in db i found such config

```
[ConfigStorage]
Items=TRX,TRX1,TRX2,ROT,ROT1,ROT2,xplanet,CW,fldigi,wsjt,Backup,WindowSize, Window,Grayline,ColumnSize,

```
Items is a comma separated list of sections to be managed locally.. But you can notice a space between windowsize and window section. I Think (not sure) that all section listed AFTER the space were ignored and probably in my initial config the Backup section was positioned AFTER the space. I removed the space and then start again the whole configuration.
Now it works as expected and cqrlog "override" db settings with local one when instructed to do so. 

So issue for me is closed, just suggest to double check for that "strange" space.

ik5pwc, 2023-10-24:
I Confirm that when there is a space some "local" options are being ignored. BTW, adjusting  local/common preferences I found another "strange" value"
```
[ConfigStorage]
Items=TRX,TRX1,TRX2,ROT,ROT1,ROT2,xplanet,CW,fldigi,wsjt,n1mmBackup,WindowSize, Window,Grayline,ColumnSize,
```
apart from space before _Window_ you notice the n1mmBackup which I think is an unknown option (probably should be n1mm,Backup?). I Modified it removing n1mm and leaving only Backup and all went back as expected.
I think there is something strange on how this value is build when configuring local/common preferences.

ik5pwc, 2023-10-24:
Tried to find how the value is build and probably found something in fConfigStorage.pas (I'm a sysadmin who sometimes plays writing code)

```
 if cmbStoreExport.ItemIndex>0 then
    Sections := Sections + 'Export,';
  if cmbStoreCluster.ItemIndex>0 then
    Sections := Sections + 'DXCluster,';
  if cmbStoreFonts.ItemIndex>0 then
    Sections := Sections + 'Fonts,';
  if cmbStoreMembership.ItemIndex>0 then
    Sections := Sections + 'Clubs,FirstClub,SecondClub,ThirdClub,FifthClub,FourthClub,';
  if cmbStoreBandMap.ItemIndex>0 then
    Sections := Sections + Sections + 'BandMap,BandMapFilter,';
  if cmbStoreXplanet.ItemIndex>0 then
    Sections := Sections + 'xplanet,';
  if cmbStoreZipCode.ItemIndex>0 then
    Sections := Sections + 'ZipCode,';
  if cmbStoreLoTW.ItemIndex>0 then
    Sections := Sections + 'LoTW,';
  if cmbStoreCWInterface.ItemIndex>0 then
    Sections := Sections + 'CW,CW1,CW2,CW3,CW4,CW5,CW6';
  if cmbStoreFldigiInterface.ItemIndex>0 then
    Sections := Sections + 'fldigi,wsjt,n1mm';
  if cmbStoreAutoBackup.ItemIndex>0 then
    Sections := Sections + 'Backup,';
  if cmbStoreExtViewers.ItemIndex>0 then
    Sections := Sections + 'ExtView,';
  if cmbStoreCalbook.ItemIndex>0 then
    Sections := Sections + 'CallBook,';
  if cmbStoreRbn.ItemIndex>0 then
    Sections := Sections + 'RBN,RBNFilter,';
  if cmbStoreOnlineLog.ItemIndex>0 then
    Sections := Sections + 'OnlineLog,';
  if cmbStoreWindowSize.ItemIndex>0 then
    Sections := Sections + 'WindowSize,Window,Grayline,';
  if cmbStoreColumnSize.ItemIndex>0 then
    Sections := Sections + 'ColumnSize,';
  cqrini.WriteString('ConfigStorage','Items',Sections);
  cqrini.LoadLocalSectionsList;
  ModalResult := mrOK

```
Based on the code above, I see that the string is build based on combobox value.
For every combo box, a value is appended. When appending, value is terminated always with a comma EXCEPT for this couple of options
```
  if cmbStoreCWInterface.ItemIndex>0 then
    Sections := Sections + 'CW,CW1,CW2,CW3,CW4,CW5,CW6';
  if cmbStoreFldigiInterface.ItemIndex>0 then
    Sections := Sections + 'fldigi,wsjt,n1mm';

```
IMHO it should be
```
  if cmbStoreCWInterface.ItemIndex>0 then
    Sections := Sections + 'CW,CW1,CW2,CW3,CW4,CW5,CW6,';
  if cmbStoreFldigiInterface.ItemIndex>0 then
    Sections := Sections + 'fldigi,wsjt,n1mm,';

```
HTH
Ciao

OH1KH, 2023-10-25:
HI!
Thanks for digging out missing commas!
Note also this push that I made yesterday to clean up Xlocal.cgf (X is the number of used log)
https://github.com/ok2cqr/cqrlog/commit/67bae0612b7f514548d0d34aa1e6335ed336dc00

OH1KH, 2023-10-25:
Commas were not only problem.
Because of the way of handling multi TRX, CW and Modes in preferences, the Xlocal.cfg may not be flushed. It must keep old settings. (this was the easiest fix for lazy programmer)
How ever when setting TRX ,CW or Modes as "Local computer only" user MUST go through also those three setting tabs .
They will appear as empty at first time after "Local computer only" is assigned.

Easy way to study changes is to `cd ~/.config/cqrlog` folder and issue:
`grep -ni -A1 "configs" 7cqrlog.cfg; grep -ni -A1 "\[" 7local.cfg`       (change "7" to used log number)
every time after preferences are closed with new settings.

https://github.com/OH1KH/cqrlog/commit/a3c93a239b84f5530127d6ce83225417cdee52da

ik5pwc, 2023-12-27:
All issues have been addressed and solver

closed 2023-12-27 by ik5pwc