ok2cqr/cqrlog #568: MacOS M1 crashes in TApplication.CreateForm

dl8dtl, 2024-05-12:
I once started a branch to port CQRlog to MacOS, in particular to the Apple Silicon version.

https://github.com/dl8dtl/cqrlog/tree/macos_port

That used to work, well, to some degree at least. Recently, I cannot get it to run at all. It crashes now with an invalid access exception in TApplication.CreateForm, at the second call to this procedure. The call stack is

```
#0 DUTILS$_$TDMUTILS_$__$$_SORTARRAY$LONGINT$LONGINT at :-1
#1 DDATA$_$TDMDATA_$__$$_LOADEQSLCALLS at :-1
#2 DDATA$_$TDMDATA_$__$$_DATAMODULECREATE$TOBJECT at :-1
#3 CLASSES$_$TDATAMODULE_$__$$_DOCREATE at :-1
#4 CLASSES$_$TDATAMODULE_$__$$_AFTERCONSTRUCTION at :-1
#5 CLASSES$_$TDATAMODULE_$__$$_CREATE$TCOMPONENT$$TDATAMODULE at :-1
#6 CREATEFORM(0x000000010480ca08, 0x0000000100aab3f0, <unavailable>) at application.inc:2176
#7 PASCALMAIN at :-1
#8 FPC_SysEntry at :-1
#9 FPC_SYSTEMMAIN at :-1
#10  at :0
#11  at :0
```

at Instance.Create(Self) (line 2176).

I have basic Pascal knowledge, but I am completely at a loss with the complexity of this.

Anyone having any clues what this might mean?

OH1KH, 2024-05-13:
Hi!
MacOS is unknown here, but what is the Lazarus version you are using?
Latest lazarus > 3 does not create working Cqrlog. With some of widget selections it does not even compile.

Well, I do not know the very latest Lazarus. I have not checked how many fix versions are released since 3.2.
Latest sure working Lazarus is 2.2.6 when using GTK2 or QT5 widgets. With GTK3 that does not compile either.

Just wondering could your problem be related to Lazarus version in use, not the Cqrlog itself.

dl8dtl, 2024-05-13:
```
% port info lazarus
lazarus @2.2.6-0 (devel)
```
So it's not latest …
I get many many pointer alignment errors/warnings during compilation, like:
```
Error: ld: warning: pointer not aligned at _$$fpclocal$_ld58+0x1C from /Users/j/.lazarus/lib/SynEdit/units/aarch64-darwin/qt5/synedit.o
Error: ld: warning: pointer not aligned at _$$fpclocal$_ld8+0x34 from /Users/j/.lazarus/lib/MemDSLaz/lib/aarch64-darwin/qt5/frmselectdataset.o
Error: ld: warning: pointer not aligned at _$$fpclocal$_ld8+0x3C from /Users/j/.lazarus/lib/MemDSLaz/lib/aarch64-darwin/qt5/frmselectdataset.o
Error: ld: warning: pointer not aligned at _$$fpclocal$_ld8+0x44 from /Users/j/.lazarus/lib/MemDSLaz/lib/aarch64-darwin/qt5/frmselectdataset.o
Error: ld: warning: pointer not aligned at _$$fpclocal$_ld8+0x4C from /Users/j/.lazarus/lib/MemDSLaz/lib/aarch64-darwin/qt5/frmselectdataset.o
Error: ld: warning: pointer not aligned at _$$fpclocal$_ld8+0x54 from /Users/j/.lazarus/lib/MemDSLaz/lib/aarch64-darwin/qt5/frmselectdataset.o
Error: ld: warning: pointer not aligned at _$$fpclocal$_ld8+0x5C from /Users/j/.lazarus/lib/MemDSLaz/lib/aarch64-darwin/qt5/frmselectdataset.o
Error: ld: warning: pointer not aligned at FPC_THREADVARTABLES+0x4 from /Users/j/src/cqrlog/src/cqrlog.o
Error: ld: warning: pointer not aligned at FPC_THREADVARTABLES+0xC from /Users/j/src/cqrlog/src/cqrlog.o
```
Not sure whether this might be related.

I have chosen Qt5 as the toolkit, since for other reasons, the ports tree is compiled without X11 support, so I cannot use Gtk (it complains about a missing library then).

The memory access violation encountered looks to me like a null pointer for some kind of struct/object, plus an offset into that one ($3F8 if I remember correctly).

I understand too little of the Pascal ecosystem used here, and who calls what. What is TApplication.CreateForm be used for? It successfully completes at the first call, but crashes at the next one.

I did successfully compile CQRlog on that device before, presumably with an older version of Lazarus and Freepascal. It's a bit sad it no longer works, because the Apple Silicon CPU is really a huge step forward compared to Intel, in particular for portable operation.

OH1KH, 2024-05-14:
Ok!
Somehow your problem has same kind of error I got when trying laz 3.2. I got it compiled  with qt5 special unit by https://github.com/davidbannon/libqt5pas.
But when Cqrlog stared it crashed just at form creation phase (or right after it finished).

My Pascal starts from 1982 and I do not have either very good knowledge of these graphical UIs like Delphi and Lazarus.
You'll find many "create forms" from cqrlog.lpr where most of used windows are created at the program beginning.

ok2cqr, 2026-10-08:
This should be already fixed because CQRLOG has working macOs port. I use it every day :).

closed 2026-10-08 by ok2cqr