ok2cqr/cqrlog #576: Update Ubuntu PPA

AsciiWolf, 2024-09-16:
Since version 2.5, CQRLOG [depends on mariadb](https://github.com/ok2cqr/cqrlog/issues/291#issuecomment-770383265) and does not work properly with original mysqld. There was no problem in Debian because it uses mariadb as its default mysql server, however it is broken in Ubuntu, (I believe) since 20.04, because Ubuntu uses mysqld as the default one.

It recently got [changed in Debian](https://salsa.debian.org/debian-hamradio-team/cqrlog/-/commit/33bad7fa4511f96d4ff48a3b868b583798ee92d6) to hard-depend on mariadb and a Stable Release Update for supported Ubuntu LTS releases was prepared.

However, Ubuntu developers have rejected the update and are [not happy](https://bugs.launchpad.net/ubuntu/+source/cqrlog/+bug/1872002/comments/20) about it so it is possible that CQRLOG in Ubuntu (and its derivatives like Linux Mint) will sadly remain broken in future releases.

Because of this, please consider at least updating the [official PPA](https://launchpad.net/~ok2cqr/+archive/ubuntu/ppa) to include releases for Ubuntu 22.04 and 24.04 to allow official installation of CQRLOG on modern Ubuntu (Linux Mint) systems.

Thanks!

ok2cqr, 2024-09-22:
I just spend maybe 15 minutes trying to purge mariadb from the Ubuntu 24.04. Removed MariaDB packages went fine.  

After a few attempts deleted:

- /etc/mysql/FROZEN
- /var/lib/mysql*
- /var/log/mysql
- /etc/mysql

Installed the mysql-server and it won't start first because of existing directories listed above. After purge mysql-server, deleted also configuration in /etc/mysql and all directories and after a new install the mysql-server wont start because it does not have any configuration. Shouldn't the install process create one? 

I don't have time for this. I'm giving up creating any packages. Ubuntu has default MySQL, Debian MariaDB. Great! Now I have to create separate deb packages for Ubuntu and Debian as well. That's crazy! Another package for Linux Mint? As far as I know, they use MariaDB as default.

I don't have energy to solve that problems. I tried to use | in dependencies MariaDB | MySQL but it does not work at all. Now, I just create a .tar.gz with binaries and step by step guide how to install it. That's all, I won't spend hours trying to solve issues in Debian, later in Ubuntu and finally with Linux Mint. It does not make any sense.  I'm giving it up.

With the first install, CQRLOG starts it's own mysql/mariadb-server instance and stores data into ~/.config/cqrlog. When I tried to do it with the mysql-server it failed with error message that I don't remember. There was no problem with MariaDB. 

It's real PITA to develop anything with GUI for Linux. So many distributions with so many versions of everything from Lazarus, FPC, MySQL/MariaDB, openssl, tqsl, trustedqsl, GTK and other stuff. I don't want to spend time with this anymore.  I'm giving it up.

AsciiWolf, 2024-09-22:
Just give up on MySQL and use MariaDB like we did in Debian:

https://salsa.debian.org/debian-hamradio-team/cqrlog/-/commit/33bad7fa4511f96d4ff48a3b868b583798ee92d6

Ubuntu is probably [moving back](https://www.phoronix.com/news/Ubuntu-25.04-Maybe-MariaDB) to MariaDB anyway (but that will not happen until 25.04).

closed 2026-10-08 by ok2cqr